opendata-stuttgart/sensors-software #836: Weird timing issue SDS011 with NRZ-2020-131/NL

pe0ter, 2020-11-25:
A few days ago I assembled 20 Airrohr units and had/have them running in one room for stability and accuracy review.
Next morning I noticed on https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&theme=light&var-chipID=
I had 2 units with irregular PM measurements, BME280 and WiFi signal were smooth and ok for all units.
One unit did measure fine every other measurement 1,3,5,7,... ok and 2,4,6,8,... no measurement data but when I changed the standard interval time from 145 to 151 sec it performed flawless ever since. After changing all my time interval settings to 151 sec I have no problems anymore.
Could there be a timing issue with the tasks the ESP8266 has to perform at 145 sec and 290 sec for the above to occur?

ricki-z, 2020-11-25:
Were all 20 devices running at the same time?
The ESP8266 must handle the wifi traffic in parallel to our firmware. So if there is many wifi traffic the NodeMCU will have a higher load. "Spreading" the transmissions by some seconds may solve this in your situation.
There are no defined tasks at 145 and 290 seconds in our firmware.

pe0ter, 2020-11-25:
@ricki-z
I had all 20 units running at the same time with a wired high-speed access-point within 1m and Wifi signal quality on all units at 100 percent.
But the unit measuring PM every other measurement is on my table downstairs and as soon as I change the time back to 145 sec it skips PM measurements again so it does not seem te be related to wifi performance issues.
I starts skipping the 'start measurement' signal to the SDS011 every other measurement when cycle is set to 145 sec.

ricki-z, 2020-11-25:
WiFi is a shared medium! The NodeMCU only supports speeds up to 54 MBit. And while one device is sending (or receiving data from the access point) all others are blocked. But they all have to listen for the packets as they need to detect the target of the packet. And this is work the EXP8266 has to do.

pe0ter, 2020-11-25:
@ricki-z 
Fully agree with you, but the interesting point is my problem moves with this specific sensor when I connect it to various NodeMCU's. If the probles was wifi related it should have been more or less random over the 20 units and it is not. If it was wifi related, changing the interval time from 145 to 151 sec would not have taken the problem away.
Anyway, let's stop this discussion: your firmware is absolutely fine, my SDS011 is working perfectly, I just have to set my timing to 151sec and all my PM sensor communicarion problems are gone and that is what I have done.

closed 2020-11-25 by pe0ter