opendata-stuttgart/sensors-software #842: Sensor number on map

jmdijkstra, 2020-12-04:
Dear readers, I am new on the subject and have a question.
Today I have constructed and installed an air measurement sensor as part of community measuring project in Holland named onzelucht.nl (our air)
I have noticed that when I check my sensor on the "Map. Sensor. Community" map, the number indicated on the map on the location where the sensor is installed is not in line with the number given to my by the supplier. The number begins with a #. 
And the location on the map is not exact the real installation location, so I am wondering whether it is my sensor of the one from someone in the neighbourhood.
Can anyone clarify this for me? Thanks in advance,
Joh. Dijkstra

pjgueno, 2020-12-05:
Hi,
In devices.sensor.community, did you check the checkbox "Publish exact location" ? Otherwise, there is a randomization in a radius of 100 m.
[image]

jmdijkstra, 2020-12-05:
Dear Pierre- Jean, thanks for the reply. I have corrected it and it's now better.
Regards, Johannes

pjgueno, 2020-12-05:
Done!

closed 2020-12-05 by pjgueno