opendata-stuttgart/sensors-software #843: DNMS T4.0+NodeMCU1.4 either DNMS or SPS30 values

box99h, 2020-12-04:
Hello,

I'm encountering an issue using my DNMS T4.0+NodeMCU1.4 PCB. The PCB is connected to a DNMS, a SPS30 and a BME280. While the BME280 is working fine I only get values either for the SPS30 or the DNMS.
If the SPS30 is connected the DNMS is not showing any values, while with a disconnected SPS30 the DNMS becomes alive.
What is especially strange is that I had onetime all sensors in operation. All sensors worked after I've changed my USB power adapter from 5W to 10W. So I assumed it might be linked to some undervoltage situation. But after a reboot the 10W is also not working with all sensors and I'm stuck in the either or situation.

I already had a similar situation with my other same config PCB at another spot where I've experienced this behavior once. But in this case a reboot solved the situation. Nevertheless the DNMS also has some hiccups on this PCB as well.

Any ideas are highly appreciated.

pjgueno, 2020-12-05:
It should basically be a problem with the power supply. I am not used to the 5W / 10W can you give me the value in mA/A ? If I remember well P = U x I so I = 220 / 10. Do you use a 22 A power supply ? It seems ti be quite high...

box99h, 2020-12-05:
I've actually tried three different power adapters (5V/2.1A, 5V/2.4A, 5V/1A). 
Meanwhile if assembled a complete new DNMS T4. Beside the Teensy 4.0 everything changes and same behavior.

pjgueno, 2020-12-05:
As both the SPS30 and rhe DNMS use the I2C, there is maybe an issue with the pins. Can you double check the connections. In the @hbitter version it works however. Do you use one of the PCB he developped ?

box99h, 2020-12-05:
It's the PCB out of @hbitter GitHub. Actually the board is working in the same configuration (beside some DNMS hiccups with extrem high LAmax readings) at another spot.

I will check all pins once again, but I'm pretty sure everything is correct. At the moment I'm running DNMS and SPS30 on two different ESP. Not perfect but it works for the moment.

hbitter, 2020-12-05:
Hi box99h,

do you spreak about a configuration like the one one the picture?
[image]

Could you check the serial debug output from the NodeMCU at restart (like the following picture) when all sensors are connected to the board?
[image]

As you can see it is running with all the connected sensors DNMS, SPS30 and BME280
[image]

Could you send me a screenshot of the actual values as well including the information about WiFi signal an quality (last two lines on that page). 

Also a screenshot of the device status, please.
[image]

A question, what values where shown for LAmax when you speak about DNMS hiccups?

Regards,

Helmut
P.S. you can send me the information as PM to helmut.bitter (at) t-online.de

linuxlurak, 2021-01-04:
I remarked "hiccups" too. Check this sensor: https://api-rrd.madavi.de/grafana/d/BYsfp-xGz/dnms?orgId=1&var-chipID=esp8266-586719

Sometimes it gets extreme values:
[image]

linuxlurak, 2021-01-04:
Here is another Sensor. This one has three "hiccups" with value 825. 
Link to the sensor: https://api-rrd.madavi.de/grafana/d/Fk6mw1WGz/wifi-signal?orgId=1&var-chipID=esp8266-7876127
[image]

There are other issues... BME280 does not deliver any data in a range that makes sense... WIFI connection seems to be of bad qualität. But I guess this is not connecte (or is it?).

box99h, 2021-01-05:
So far everything points to the bad wifi connection. 
I had my station for about one week in-house in proximity to the Access Point and the hiccups were gone. Right after putting the station back outdoor (with bad wifi signal) the hiccups were back.

As far as I understood the problem is caused by the ESP which is jamming the Teensy sometimes in bad wifi environments.
Helmut and his teams are aware of this issue and are checking how to resolve. 

After weighing up all the pros and cons I will change my station to the Airrohr V1.4 to increase distance between ESP and Teensy.

linuxlurak, 2021-01-05:
Thanks for the info. Do you have a link to an issue by any chance?

hbitter, 2021-01-31:
Two different things are addressed:

- Peaks/hiccups with noise values > 130 dBA and/or noise values = 0
  This seems to be caused by interference between WiFi and Teensy4.0 in case of a bad WiFi connection. In short time frame    there will be a new PCB version (DNMS T4.0+NodeMCU1.5) available. The new version will have "ground-planes" to lower the interference of the signals.

- Either DNMS or SPS30
  With newer Teensy4.0 boards simultaneous operation of DNMS and SPS30 does not work. On my test system with an older Teensy4.0 it does however work. On the working Teensy4.0 the i.MX RT1062 processor had the mark "C1AA1848H" in the 4. row. On Teensy4.0 boards, that did not work, the i.MX RT1062 processor has the mark "CTBY1947B" or "CTBU1942B". It needs further investigation. In the meantime the workaround is using separated Airrohr PCB and DNMS-T4.0 PCB instead.

Franz4596, 2021-02-22:
1. @ pjgueno commented on 5 Dec 2020
"It should basically be a problem with the power supply. I am not used to the 5W / 10W can you give me the value in mA/A ? If I remember well P = U x I so I = 220 / 10. Do you use a 22 A power supply ? It seems ti be quite high..."
-> right calculation is:  I=P/U =10W/22V=45mA or 10W/5V=2A  (not 22A)
-> use of  USB 5V / 2A suppy is recommended   (5V/500mA may not work)

2. issue BME280:  sometimes the chinese BME280 hang up to -150°C, software reboot usuall not helps, you need a power off/on. If this not solve the fault the BME280 has be replaced. Probably a protectiv farnish helps.

3. Maybe the stored DNMS dataset at madavi.de and sensor.community are  different?

4. DNMS T4.0+NodeMCU-V1.5 PCB has DNMS-SPS30 fault issue  sometimes too, 
    DNMS T4.0+NodeMCU-V1.4 sometimes work 
 in my surrounding. See esp8266-14726029: DNMS-T4.0+NodeMCU-V1.4, SPS30, SHT31