opendata-stuttgart/sensors-software #844: How to compare graphs and trends of PM sensors in neighbourhood

jmdijkstra, 2020-12-06:
Hello, I have noticed quite some differences in PM date measured by myself and PM data measured by a sensor in the neighbourhood of my sensor. In order to further investigate I would like to compare graphs and trends taken from the website PM Sensors Grafana (https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&theme=light&var-chipID=esp8266-116455) However I need a sensor ID in order to select the neighbourhood sensor. Can anyone tell my how I can recover these sensor information?

Thanks in advance,
Joh. Dijkstra
p.s. is this the only way to discuss issues about the www.onzelucht.nl project?

bertrik, 2020-12-06:
IMO, this issue tracker is for technical issues related to the firmware.

There is a sensor.community chat platform at
  https://chat.sensor.community/
Also I think a forum should come online soon at:
  https://forum.sensor.community/
Historical data is available at:
  http://archive.sensor.community/

With respect to ids, there are several:
* A kind of "hardware id", derived from the unique number in the ESP8266 microcontroller in each kit. I *think* this is what madavi uses.
* A sensor.community database id. Actually there are typically two, one for the PM-sensor and another one (usually PM-sensor-id + 1) for the meteorological data (temp, humidity, etc). This is the id that you see if you look up your sensor on the maps.sensor.community map
* If you look at the samenmeten kaart, there is another id, typically LTD_XXXX, the XXX is the sensor.community the *location id"
This can be found at https://samenmeten.rivm.nl/dataportaal/

There are several groups in the Netherlands doing comparisons of sensor data, e.g. making cross-correlation diagrams to see if sensor kits in general agree with each other, and to see if there are kits that are acting peculiar, which might be an indication of a sensor failure or an actual local pollution source. Also doing correlations between wind direction and intensity vs PM values. The group in Leidschendam is very active with this. See this link for an e-mail address
https://vlietnieuws.nl/2020/01/13/u-zegt-het-maar-luchtkwaliteit-bij-de-jaarwisseling-was-slecht-tot-zeer-slecht/

ropeters68, 2020-12-07:
NL users can download hourly data from https://samenmeten.rivm.nl/dataportaal/. Just select a sensor, the nice part is that you can enter a start and end date for which you want to download the values so you don't have to combine several files and besides the raw data you can download the values with humidity correction too (under current conditions you're sensor readings will be too high, see screen shot, download by selecting the button in the green circle) and the data from the nearest reference sensor. Looks like you're using SDS011 sensors? I would ignore PM10 reading as these are not reliable. 
[image]

jmdijkstra, 2020-12-07:
Hello ropeters68 and bertrik, 
Thanks for the feed back, clarifies a lot.
With best regards,
jmdijkstra

closed 2020-12-07 by jmdijkstra