opendata-stuttgart/sensors-software #846: docs: document the supported SSL ciphers

bb-Ricardo, 2020-12-07:
I updated my sensors from a old firmware and after that the influx SSL writer stopped working. After some poking around I got a 'SSL cipher mismatch'.

It took me a while to figure out which ciphers are actually supported. To make sure people know that only a basic set of ciphers are supported it should be documented.

The common build_flags in platformio.ini contain `-D BEARSSL_SSL_BASIC` which limits the set to
```
    BR_TLS_RSA_WITH_AES_128_CBC_SHA256,
    BR_TLS_RSA_WITH_AES_256_CBC_SHA256,
    BR_TLS_RSA_WITH_AES_128_CBC_SHA,
    BR_TLS_RSA_WITH_AES_256_CBC_SHA,
```
CBC ciphers are commonly considered insecure. But I guess due to the limited resources this is the best combination of encryption and performance.

It would be great if this could be added as a small note/hint in the docs. Or it is already documented and I didn't find it.

Thank you, for all this work. This is an amazing project.

ricki-z, 2020-12-08:
Added to documentation at https://github.com/opendata-stuttgart/meta/wiki/APIs
English version will follow.

bb-Ricardo, 2020-12-08:
great, thank you very much

DeeKey, 2020-12-25:
@ricki-z merged both German and English API pages. Also added 3rd party API description:
https://github.com/opendata-stuttgart/meta/wiki/EN-APIs