opendata-stuttgart/sensors-software #848: humidity sensor at 99.9%

alexdelli, 2020-12-10:
Good morning
Why the humidity sensor is even at 99.9% ?
How can I check where is the problem? I don't know when it starts this issue.

Thank you

GitHub4Eddy, 2020-12-10:
Is it the DHT22 sensor?

If so:

The DHT22 sensor is originally an Indoor sensor. For outdoor use it appears to be rather sensitive to water condensation after 100% rel.Humidity that keep it for very long time (sometimes forever) at 99.9% value. Also it appears to be sensitive to high UV light, which tends to cause the sensor to crash until hard power-loss restarted. Better experiences have been made with a BME280 or SHT3x sensor. So if you use a DHT22, consider to replace it with a BME280 instead. BTW The BME280 sensor also measures air pressure and costs less than 2 euro.

alexdelli, 2020-12-14:
> Is it the DHT22 sensor?
>
> If so:
>
> The DHT22 sensor is originally an Indoor sensor. For outdoor use it appears to be rather sensitive to water condensation after 100% rel.Humidity that keep it for [...]

Yes is that sensor. I purchased a new BME280. Thank you for your help

closed 2020-12-14 by alexdelli