opendata-stuttgart/sensors-software #849: Comparing two sensors close to each other

jmdijkstra, 2020-12-11:
Hello, In order to get an idea about the accuracy of my sensor I have downloaded from the site Samen meten (RIVM) data from my sensor and data from a sensor 200 metres away  and compared theses data in the same period of time in a excel graph and noticed quite some differences . My sensor readings are sometimes twice the readings of the other sensor. Can anyone explain this because I can hardly imagen that air quality can that much differ.

RikDrabs, 2020-12-11:
"Hallo de buren, de groeten in Nederland!"

What are your measurements? Do you measure a lot more, or a lot less than your neighbor? And which values exactly? Can you give me your chip-id, and the api-id of the neighbor in the comparison, so that i can see for myself? 

Data can diverge a lot, depending on the distance between the two, local polluters burning wood, and differences in the SDS011 sensor. When placed side by side two sensors can already diverge for a number of µg/m³, even when completely OK. Nova, the maker of the sensor, specifies an accuracy of +/-10%. This means that one sensor can be at +10%, and another at -10%. The difference between the two can thus be as high as 20% with tested good sensors, and the same pollution level. Of course there is always the possibility that a sensor is malfunctioning. Look around on the map of Belgium, and notice that sometimes one sensor "stands out" and gives a much too high value. (> 100 µg / m³) while its neighbors give normal values. You can see this by the lonely red or purple color of some sensors. These sensors are mostly defective. When you replace that SDS011 with a new one, everything will be fine again. I did it already multiple times. On one occasion i had an SDS011 that gave results 10 times too high, and returned to normal low values in case of no pollution. I replaced it, and everything returned to normal. People burning wood can also upset their neighbors sensor a lot, without influencing sensors at large. You could be in that situation. I wish you luck convincing your neighbor to stop burning wood, if this is your case ...

mvg,
Rik,
Support info@luchtpijp.be - Belgium

jmdijkstra, 2020-12-11:
ID: 116455 (10521c01c6e7) Dit is volgens mij mijn chip-id?

Verzonden vanuit Mail voor Windows 10

Van: Rik Drabs
Verzonden: vrijdag 11 december 2020 18:08
Aan: opendata-stuttgart/sensors-software
CC: jmdijkstra; Author
Onderwerp: Re: [opendata-stuttgart/sensors-software] Comparing two sensors closeto each other (#849)

"Hallo de buren, de groeten in Nederland!"
What are your measurements? Do you measure a lot more, or a lot less than your neighbor? And which values exactly? Can you give me your chip-id, and the api-id of the neighbor in the comparison, so that i can see for myself?
Data can diverge a lot, depending on the distance between the two, local polluters burning wood, and differences in the SDS011 sensor. When placed side by side two sensors can already diverge for a number of µg/m³, even when completely OK. Nova, the maker of the sensor, specifies an accuracy of +/-10%. This means that one sensor can be at +10%, and another at -10%. The difference between the two can thus be as high as 20% with tested good sensors, and the same pollution level. Of course there is always the possibility that a sensor is malfunctioning. Look around on the map of Belgium, and notice that sometimes one sensor "stands out" and gives a much too high value. (> 100 µg / m³) while its neighbors give normal values. You can see this by the lonely red or purple color of some sensors. These sensors are mostly defective. When you replace that SDS011 with a new one, everything will be fine again. I did it already multiple times. On one occasion i had an SDS011 that gave results 10 times too high, and returned to normal low values in case of no pollution. I replaced it, and everything returned to normal. People burning wood can also upset their neighbors sensor a lot, without influencing sensors at large. You could be in that situation. I wish you luck convincing your neighbor to stop burning wood, if this is your case ...
mvg,
Rik,
Support info@luchtpijp.be - Belgium

ropeters68, 2020-12-11:
If the sensors are only 200m apart, readings should be similar, hyper local pollution gives irregular high peaks I would not expect structural differences. I would guess the other sensor might be installed indoor or has an issue with airflow (inlet or outlet). In the samen meten portal you can select the nearest reference sensor. You can check which of the sensors matches best. A more reliable sensor is the Sensirion sps30 (unfortunately also more expensive).