opendata-stuttgart/sensors-software #850: Pariculate sensor SDS011: No wireless connection

githubsba, 2020-12-11:
Via my computer(wired connection with wireless-modem) 
I get the data of my pariculate sensor SDS011  at airrohr-XXXXXXXX.local .

But via my phone or tablet I don't get it.
The browser says: "Can't find it"

What could be the cause here?

RikDrabs, 2020-12-11:
You're sure that the phone is connected to your local WiFi? The address you're using does not work over the internet at large, because this is the local address, neccessitating a local connection between sensor and the machine running the browser, over WiFi, cable, etc. The correct address to use over internet is: 
https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&theme=light&var-chipID=esp8266-XXXXXXXX#. This works at home, but also when not at home.

githubsba, 2020-12-11:
RikDrabs, thanks for your quick response. Appreciate that.

Url: api-rrd... 
The large url starting with api-rrd... works fine via my computer.
 I see nice graphs.
But not on the phone. It says "Dashboard not found".

Local wifi:
Checking my wifi on the phone: the chip SSID is not there.
When I add it, it doesn't do anything.

Chip:
So, while the chip has good internet connection via my wifi network, I can't approach it with my mobile devices.

Any suggestions?

RikDrabs, 2020-12-11:
Chip SSID is not visible after a correct install on your home WiFi. This is normal.
Readout of graphic results is always trough the server, with the large url api-rdd...
Does the browser you're using on the phone & on the tablet supports Grafana? Your computer clearly does. But the error message on phone & tablet "Dashboard not found" suggests that the problem there is with Grafana. This is the program supporting the local drawing of graphics. Maybe your browser there is not compatible. Use a more recent "Chrome" browser on phone & tablet, if your devices are older.

@ricki-z : do you know the answer here, if the problem persists?

githubsba, 2020-12-11:
Update: 
Got your first response also on the phone.
When I clicked the long the api-rrd link, and filled in my chip number (without the #)
then indeed I can see the sensor graphs on my phone.

Thanks again for your help.

githubsba

ricki-z, 2020-12-11:
Hello,
if it's an Android phone then you can't open the .local address. Android doesn't support the Bonjour protocol used for these addresses. But there are apps to scan for Bonjour enabled devices. And there is an app especially for our sensors:
https://play.google.com/store/apps/details?id=net.liutikas.sensormanager

githubsba, 2020-12-15:
Accessing devices with a .local address, on android devices.

Too bad you need an app for that.

Anyway, ricki-z thanks for your great tip.

I installed the app from Aurimas Liutikas, 
and can now view de data of my sensor on my phone. 
The app works nicely.

ricki-z, 2020-12-15:
Maybe Android will support this in newer versions.
And it's only the name resolution. If you can get the IP address of the sensor (most routers have a list of connected devices that also shows the IP address) you can open this IP address in a browser.
The name 'feinstaubsensor-xxxx.local' (or 'airrohr-xxxx.local' for newer devices) is easier to remember and doesn't change while an IP address received via DHCP may change over time.

githubsba, 2020-12-17:
Even better so. I used the ip address of my internet-modem,
and there is indeed the list of connected devices with their ip addresses etc. Tried the ip address of the nodemcu, and after a few attempts I could see the sensor data on my android device.

Thanks again for your help.

ricki-z, 2021-04-02:
@githubsba can we close this issue?

githubsba, 2021-04-02:
Yes, you can close this issue.

Thanks,

Sybren.

closed 2021-04-02 by ricki-z