opendata-stuttgart/sensors-software #852: Problem connecting to wifi router

git-user44, 2020-12-14:
I'm just starting to use this software, and working out what it is capable of, but I came across a problem.

After I input my router SSID and password  for the first time, it would never connect. I have traced the problem in connectWifi() to the line ...

	wifi.nchan = (INTL_LANG[0] == 'E' && INTL_LANG[1] == 'N') ? 11 : 13;

My router runs on channel 13. I am in the United Kingdom. 13 channels are valid here because we (still) follow Europe in this regard. 11 is the valid number for the USA.

I changed my source to wifi.nchan = 13; and the software burst into life.

I cannot see a simple solution to this unless you include some sort of regulatory authority/geographical indicator.

DeeKey, 2021-01-05:
Maybe it make sense to add server based checkup for location?
If channel restriction is set, NodeMCU continue to other steps.
If channel restriction is not set, then NodeMCU asks from server for valid restrictions (store them to configuration).
If NodeMCU is not yet registered on the server then warning message "please register on device.sensor.community" and apply all available world channel restriction (as we do not know where the device is).

Also in this case one first need to register device and only then configure in local network.

ricki-z, 2021-01-05:
@DeeKey server based checkup won't work. How should the sensor access this without a working wifi connection?

This problem is caused by a "mistake" in the program logic. We only need to limit the channels while the sensor is opening the hotspot for configuration. If the device is in client mode then it can scan on all channels for available wifi networks (in countries with a limitation to 11 channels there will be no network on channels above channel 11).
So moving the mentioned line to the part of the source where the hotspot is started should solve this problem (plus adding a line with 'wifi.nchan = 13;' to the start of the client mode).

ricki-z, 2021-04-03:
@git-user44 could you please try the latest beta? I have switched the available channels between AP mode (now 11) and station mode (now 13). This should solve this.

git-user44, 2021-04-08:
OK. Tested this on something that is not my 'live' system, just a spare ESP8266 (Wemos Mini d1)
Took existing NRZ-2020-133/EN source code, and patched it to include line  wifi.nchan=13;
It connected to my router OK, and this is what happened...

19:14:21.391 -> ⸮⸮⸮ԅ⸮@H⸮airRohr: NRZ-2020-133/EN
19:14:21.556 -> mounting FS...
19:14:21.589 -> opened config file...
19:14:21.589 -> parsed json...
19:14:21.622 -> output debug text to displays...
19:14:21.722 -> Connecting to <name of router here>
19:14:22.185 -> ..........
19:14:27.581 -> WiFi connected, IP is: 192.168.1.49
19:14:27.614 -> Starting Webserver... 192.168.1.49
19:14:27.647 -> 
19:14:27.647 -> ChipId: 8168766
19:14:27.647 -> 
19:14:27.647 -> MAC Id: <MAC ID here>
19:14:27.845 -> SNTP synced: Thu Apr  8 18:14:27 2021
19:14:27.879 -> 
19:14:27.879 -> twoStageOTAUpdate
19:14:30.096 -> Update md5: f9b529a088988d2a06dd986eb960fb3c
19:14:30.129 -> Sketch md5: 93245cf75e0bd71534bf8b86154ce43b
19:14:55.349 -> Success downloading: /airrohr/beta/latest_en.bin
19:14:56.441 -> Success downloading: /airrohr/beta/latest_en.bin.md5
19:15:10.441 -> Success downloading: /airrohr/loader-002.bin
19:15:11.765 -> launching 2nd stage
19:15:14.876 -> Erasing SDK config.
19:15:15.141 -> Restart.
19:15:15.737 -> ⸮⸮⸮ԅ⸮⸮D⸮
⸮⸮No Firmware file found, looking for: /firmware.bin
19:15:51.181 -> Update Failed.
19:15:51.214 -> Please upload the latest firmware as '/firmware.bin' on SPIFSS to recover.
19:15:59.555 -> No Firmware file found, looking for: /firmware.old
19:15:59.588 -> Update Failed.
19:15:59.621 -> Please upload the latest firmware as '/firmware.bin' on SPIFSS to recover.
19:16:07.962 -> No Firmware file found, looking for: /firmware.bin
19:16:07.995 -> Update Failed.
19:16:08.029 -> Please upload the latest firmware as '/firmware.bin' on SPIFSS to recover.
19:16:16.369 -> No Firmware file found, looking for: /firmware.old
19:16:16.402 -> Update Failed.
19:16:16.436 -> Please upload the latest firmware as '/firmware.bin' on SPIFSS to recover.
19:16:24.778 -> No Firmware file found, looking for: /firmware.bin
19:16:24.811 -> Update Failed.

and so on....

DeeKey, 2021-04-08:
I am not sure that Wemos Mini is supported hardware...

ricki-z, 2021-04-08:
This looks like some wrong compile settings (i.e. start address of SPIFFS). Please use the flashing tool, flash the normal firmware file and switch to beta version on the configuration page.

git-user44, 2021-04-09:
Sorry to be a pain, but you are going to have to make this a bit more idiot proof. Although I've been in software development for over 40 years, I've never done anything with Python, so don't understand about how it works and don't have the time to mess about with it. What is the Python "from" directive all about ? Where does it look ? Because at the moment your flasher dies.

I need to make your flasher find my esptools I guess. How ? Tried a symlink and it didn't work.

[mike@fan_brycheiniog airrohr-firmware-flasher-master]$ ./airrohr-flasher.py
Traceback (most recent call last):
  File "./airrohr-flasher.py", line 13, in <module>
    from esptool import ESPLoader, erase_flash
ModuleNotFoundError: No module named 'esptool'

pjgueno, 2021-04-09:
Why do you have to program in python to use the flasher ? it is an app you can download (look on the sensor.community site in the instruction) for mac, linux and windows. You just have to plug the board (without any component) on an usb plug. The flasher will recognize it. Then you select the version of the firmware you want. 
Follow the instructions to connect the flasher with your wifi credentials as described in the instruction. Then click on the box "use beta" and restart the firmware with the button. The beta firmware will be automatically installed. Of course you have to be let the board connected to the internet.
Another solution : clone the beta from the repo and use platformio to flash it in the board. But it will be more difficult...

git-user44, 2021-04-09:
So where is this program for Unix ? Please give me a URL or something because all I can find is source code that needs building.

pjgueno, 2021-04-09:
https://github.com/opendata-stuttgart/airrohr-firmware-flasher
https://github.com/opendata-stuttgart/airrohr-firmware-flasher/releases

pjgueno, 2021-04-09:
I already build for Unix and even succeed to make it work perfectly on my mac

ricki-z, 2021-04-09:
or "the hard way" (CLI), installing the latest beta:
download the firmware binary from https://firmware.sensor.community/airrohr/beta/latest.bin
search for esptool.py and use the following command line
<path_to_esptool.py/esptool.py  --chip esp8266 --port <path_to_usb_device> write_flash 0x0 <path_to_downloaded_latest.bin>

git-user44, 2021-04-09:
Sorry. I am giving up on this.

I found a Windows based laptop and downloaded the pre-built flasher. It flashes the chip OK, brings up an AP, I put in my router details, but of course it doesn't connect because the router is on channel 13. Tried resetting the chip a few times just in case of a noisy connection but would not connect. So no beta to download and try..

I tried it the "hard way". The chip got programmed. It brings up an AP to connect to with the airRohr-nnnnnnn name, but it is a secure system needing a password. Had a few possible guesses. Could not log on.

So I'm thinking totally wipe the chip and start again, in case some of the EEPROM is corrupted

Went back to the Arduino IDE, and told it to wipe the whole flash. Then reloaded the beta software "the hard way". It still brings up the locked AP.

Load a few test programs into the chip to prove to myself the chip is still OK.

My live sensor is solar powered. I've hacked about a version of NRZ-2020-133/EN a lot to include INA219 monitors on the batteries and solar panel, and it also talks to an arduino nano that controls two motors so that the solar panel tracks the sun. If I load that software it works. First time it brings up an access point,  I fill the details, reboot. No problems.

So I'm giving up trying to test this beta software, either I cannot download it, or it brings up a locked AP

ricki-z, 2021-04-09:
Sorry my fault. A fresh installation of the firmware has set an AP password. The password is 'airrohrcfg'.

git-user44, 2021-04-09:
OK. I can confirm this software connects to my router NRZ-2020-134-B3/EN (Apr 8 2021)