opendata-stuttgart/sensors-software #858: Incorrect readings with SDS011 with NRZ-2020-133/EN

ramonitor, 2020-12-26:
I'm getting incorrect readings with the SDS011 and NodeMCU v3 running NRZ-2020-133:

SDS011 | PM2.5 | 230.8 µg/m³
SDS011 | PM10 | 248.2 µg/m³

This is a brand new setup btw. I've tested the SDS011 with the Python script found [here](https://github.com/binh-bk/SDS011) and that shows quite 'normal' values for pm2.5 and pm10 (4.7 and 10.1 for pm2.5 and pm10 resp).

Any thoughts on this?

ramonitor, 2020-12-28:
Just tested with NRZ-2020-134-B1/EN (beta firmware); same results -> extremely high pm2.5 and pm10 values.

ricki-z, 2020-12-31:
Could you try another USB wall charger? Some are limiting the current to 100 mA. Or you connect your setup (NodeMCU + SDS011) to the same USB port that you used for your test with the script.

ramonitor, 2020-12-31:
@ricki-z ; I did test with the USB port used to test the SDS011 with the Python script, that did not solve things. I also did test with several different USB chargers. Currently using a 5v 2a output charger (used for Raspberry PI), same result; extreme high reading.

ricki-z, 2020-12-31:
can you send pictures of your setup?

ramonitor, 2021-01-01:
[image]
[image]

ricki-z, 2021-01-05:
Do you have another NodeMCU module? The high readings are normally caused by a low current or voltage. Maybe the voltage converter is defect.

ramonitor, 2021-01-06:
Thanks, I've ordered another NodeMCU module to test.

ramonitor, 2021-01-23:
@ricki-z ; Just attached a brand new NodeMCU v3, flashed the NRZ-2020-133/EN firmware. Unfortunately still the same issue. Any suggestions how to troubleshoot this further?

These are the SDS011 readings I get back:

SDS011 | PM2.5 | 176.2 µg/m³
SDS011 | PM10 | 190.8 µg/m³

RikDrabs, 2021-01-23:
Is this a brand new SDS011? 
I had the same problems with a couple SDS01's, in use since some time, that started giving too high values, some even hundreds of µgrams too high. In most cases there was a zero-level plateau from 100 to 200 µgrams, with neighbouring sensors at near zero PM2.5 and near zero PM10. 
I even had a brand new SDS011, measuring 10 times the real values, again according to neighbouring sensors. 

Replacing the SDS011 always solved the issue. 
Did you try to replace the SDS011?
Contact me directly at info@luchtpijp.be or see our website www.luchtpijp.be.

lucahammer, 2021-04-10:
I am having the same issue. New setup, SDS011 values are way too high. Switching to beta version and changing chargers and cables didn't help. I already ordered another SDS011 to see if that's the issue.

[image]

ramonitor, 2021-04-10:
I solved it by replacing the SDS011 with a brand new one. It appears not to be related to this project btw, I tried flashing the  the NodeMCU (of the appearently broken SDS011 setup) with [esphome](https://esphome.io/index.html) and had the same result. I kept the broken SDS011 for a rainy day. Maybe it will be useful somehow.

lucahammer, 2021-04-23:
The new sensor arrived and solved my problem as well. I am attaching an image of both of them. The broken one is on the left. It has 200413 A-B written on the sticker. Working one 201210 A-H. Broken one has more of a shiny finish and some additional imprints, working one a matte finish. Maybe that info helps someone with troubleshooting.

[image]

[image]