opendata-stuttgart/sensors-software #861: DHT22 Temperatur Messung wird nicht ausgewertet

zynic, 2021-01-03:
Einheit 5677653 (cc50e356a255) misst Feinstaub und mit DHT22 Temperatur und Feuchtigkeit. In https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-5677653 wird nur Feuchtigkeit ausgewertet/dargestellt. Messreihen für Temperature fehlen. Zeitweise waren -last 24 hours- sichtbar.

ricki-z, 2021-01-03:
Hallo,

ich sehe in der ersten Spalte auch Temperaturwerte. Diese sind nur recht extrem. Entweder hat eines der Kabel nur schlecht Kontakt. Das Auslesen der Daten vom DHT22 ist, weil es nur über einen Anschluß erfolgt, recht kritisch was den richtigen Kontakt der Kabel angeht.

Oder der DHT22 hat einen Defekt (weil nur die Temperatur so extrem ist).

zynic, 2021-01-03:
Hi,Danke, dann prüfe ich dies. Bleib gsondUlrich