opendata-stuttgart/sensors-software #863: time not synced with ntp servers

unsermanninchina, 2021-01-08:
Hi! 

I have two airrohre and both are not synced with ntp. All computers in the network can connect and are synced. What can I do? I've done restarts but it doesn't help. Because of the missing time - both have the date of 1.1.1970 - they can't upload to madavi and the other API's. Last upload was at the 16th of December.

Actually I updated my router, a Turris Omnia, about that time. The problem could be the router as well. Is there a way to check if the airrohr software tries to contact the ntp servers? I can sucessfully ping 0.pool.ntp.org from my laptop.

Thank you!

unsermanninchina, 2021-01-08:
[image]
[image]

DeeKey, 2021-01-10:
This issue is being discussed on the forum here:
https://forum.sensor.community/t/unable-to-sync-time-with-ntp-servers/191

unsermanninchina, 2021-01-11:
Not solved now but closing the issue here.
See [at forum.sensor.community](https://forum.sensor.community/t/unable-to-sync-time-with-ntp-servers/191)

unsermanninchina, 2021-01-21:
I have rolled back the system on my router to an earlier state and now both sensors get ntp time again. The NTP settings are exactly the same. I still don’t know what was the problem at the moment. However, I close this topic, as it seems not related with the airrohr software.

closed 2021-01-21 by unsermanninchina