opendata-stuttgart/sensors-software #864: Freifunk network withour PSW not working

odon4t4, 2021-01-11:
I've finally installed a Freifunk router and wanted to switch the used network of my sensor node...

several times I've tried to "klick" on the Freifunk network shown in the [http://192.168.4.1/config](http://192.168.4.1/config) but the sensor node was not accepting the network. Then I've switched off the Freifunk router and used my Phone to open a WiFi network with the name Freifunk AND a generic Password... all of a sudden it worked...

to be honest it was to cold to further debug the behavior so does anyone has a idea?

pjgueno, 2021-01-12:
What kind of router is it ? A vodafone router of mine was not able to support ESP8266 and I had to install another wifi router plugged on it. Is there a kind of WPS option checked?

odon4t4, 2021-01-12:
It is a TP-Link TL-WR841N/ND v9...
I do not have the time right now to test it with a different SSID and no PW on the router.

pjgueno, 2021-01-12:
The TP-Link TL-WR940N N450  ist the router I installed a few months ago to connect my sensors. It should work with this brand.It should be an option somewhere.

odon4t4, 2021-01-13:
> The TP-Link TL-WR940N N450 ist the router I installed a few months ago to connect my sensors. It should work with this brand.It should be an option somewhere.

Basically the SW defines all the features and I know that the router is a little older. I'm running a openWRT clone from the Freifunk Stuttgart guys:

2.0+2020-09-26-g.8547bd43-s.f16f34e / gluon-v2020.2.1-ffs0.1