opendata-stuttgart/sensors-software #865: DNMS T4.0+NodeMCU1.4 either DNMS or SDS011 values

JaKuRe, 2021-01-12:
Hello,

I have created a PCB based on the "DNMS T4.0+NodeMCU1.4 PCB".
Basic differences are its single sided layout and connectors for the sensors.

Everything works fine as long as no microphone is connected.

Connected devices: DHT22, BME280, SDS011, Mic.

When SDS011 is connected together with the microphone, there is an error "DNMS not found". SDS011 ist working fine.
With disconnected SDS011, the Mic is found and works fine. (tried with two different Mic-models, same issue)

During the tests, the Teensy was left in his socket and both sensors were selected in the sensor configuration at all times.
After switching off power, the connectors were inserted or removed, then put back on power.

Teensy and NodeMCU are about 2cm apart.

Any suggestions?

Thanks in advance.
Kind regards.

pjgueno, 2021-01-12:
Do the PSU provide enough current ? How many mA can it provide ?

JaKuRe, 2021-01-18:
Hello pjgueno.

Thank you for your advice.
Right away I switched the initial 1000mA PSU against a 2000mA PSU. 
I kept it running like this for almost a week and the problem is gone,

Now, the sensor gets powered by 2500mA for Outdoor use.

Kind regards :)

closed 2021-01-18 by JaKuRe

pjgueno, 2021-01-18:
You're welcome ! Where is your sensor ?

JaKuRe, 2021-01-18:
My sensor is placed in Saarland, in the valley of the Moselle.

pjgueno, 2021-01-18:
It will be the second in Saarland (with mine) !