opendata-stuttgart/sensors-software #867: Support for Influx2

9lstr, 2021-01-19:
Is there a way to send data to an Influx 2.0 database? There is a new auth process that uses a token instead of username and password. I have not found a solution in the current software for authorization with the Influx token.

m-bucher, 2021-02-26:
So basically InfluxDB2 has a 1.x compatibility mode, which suggests to use the username as username and the Token as Password: https://docs.influxdata.com/influxdb/v2.0/reference/api/influxdb-1x/#basic-authentication

Looking at the code, it seems that the maximum length a password can have is 64bytes: https://github.com/opendata-stuttgart/sensors-software/blob/b78aa0a059da6c867f5dc75f29dc86fa112808b9/airrohr-firmware/defines.h#L14

The catch is that the InfluxDB2 Tokens have 88bytes :slightly_frowning_face:

ricki-z, 2021-04-01:
Hello @9lstr , @m-bucher
The new beta firmware should have a longer password field. So the compatibility mode should work now. Could you please test this?

9lstr, 2021-04-06:
Hello @ricki-z,
For me the mentioned documentation is not clear. So I tried several combinations of user and password and none of them worked. 
There is definitely one more change in the GUI necessary. The input field has a 'maxlength="64"' definition in the html. Can you change that? Afterwards I will test it again.

ricki-z, 2021-04-08:
The latest beta (published yesterday in the evening) should have a larger input field.

apuls, 2021-06-13:
If you want to use the V2 API you will need to send a "Authorization" header via Post and some other stuff.
But you can still use the V1 authentifcation with Inlfux V2.

- Create a bucket
- Create an V1 user
- Create a Retention Policy
`$ influx v1 auth create --username 'username' --write-bucket 'bucket-id' --password 'password' --org 'Org Name'`
`$ influx v1 dbrp create --bucket-id (from above) --db feinstaub --org 'Org Name' --default --rp feinstaub`

I'm not realy sure about the '--rp' flag if it's just a name or something else....

ituri, 2022-03-15:
> If you want to use the V2 API you will need to send a "Authorization" header via Post and some other stuff. (Would be neat if the will work out of the box :D ) But you can still use the V1 authentifcation with Inlfux V2.
> 
> * Create a bucket
> * Create an V1 user
> * Create a Retention Policy
>   `$ influx v1 auth create --username 'username' --write-bucket 'bucket-id' --password 'password' --org 'Org Name'`
>   `$ influx v1 dbrp create --bucket-id (from above) --db feinstaub --org 'Org Name' --default --rp feinstaub`
> 
> I'm not realy sure about the '--rp' flag if it's just a name or something else....

I've tried to follow these steps but it doesn't really work for me. Since there is no real debug log on AirRohr's side, I am also not able to find the reason for why there is no data written to my influx2 instance.
How did you set up your configuration on AirRohr's side? Can you post a screenshot of your settings?

ituri, 2022-04-23:
I finally figured it out. The steps that @apuls mentioned are correct, however I was lacking the correct path. In my (default?) configuration it said `/query?db=iot`, which doesn't work. The path has to be `/write?db=iot` instead.
[image]

However, I still think that Influx2 support would be a good thing since I don't think that Influx 2.x will have backwards compatibility to 1.x for forever.

chevdor, 2024-03-26:
Native Influx v2 would be nice indeed.

mvoelkl, 2024-11-01:
Can confim that writing to an InfluxDB2 works with the firmware NRZ-2024-135/DE (April 2024). It even continues the InfluxDB1 dataset (renamed it from luftdaten/autogen to luftdaten after InfluxDB auto-migration; see their docs).

If writing doesn‘t work take a look at „Gerätestatus“ of your Airrohr, if in the „Fehler“ section beneath InfluxDB appears an „connection refused“ (don‘t remember the exact term) your InfluxDB is not reachable and an „401“ obviously indicates that there‘s something wrong with your InfluxDB API token or its permissions.

Used the following settings (whereas password is the looong API token and user the one you created at setup):
[image]

Should we document this somewhere as InfluxDB 1.x is gonna be disappearing gradually?