opendata-stuttgart/sensors-software #869: No SDS011 values on Serial

Beppi4U, 2021-02-02:
Debug level is set to zero
Measurement time is 120sec.
Serial connection is established at 9600 and values coming(csv style) from the BME280. No Wifi is used!
But no measurements values from the SDS011....
I use latest original firmware from Luftdaten.de
Is this because there is only one serial connection is used for the SDS011??
Any suggestions to get around this??

Hardware: ESP8266+breakoutboard : 4x20LCD : SDS011 : BME280

pjgueno, 2021-02-03:
Do you mean the latest .bin @Sensor.Community ?
Did you set up the sensor types (SDS011 and BME280) with the checkboxes correctly ?
Did you read the last version of the official instructions on the site https://sensor.community ?

Beppi4U, 2021-02-06:
In normal use i can see PM values on the LCD screen and online via web. So the wiring and Sensor seem to work fine! 
I have to double check the settings on monday when i'm on work!