opendata-stuttgart/sensors-software #872: CSV API Option

Kreid4, 2021-02-06:
Hello 
I am fairly new sensor owner.  I am wondering how the CSV API option on the device configuration page works .  Does this mean you can output data/info to a local file? 

Our sensos
https://maps.sensor.community/#14/51.0907/-1.7848

All help, would be appreciated please.
Thank  you

Regards
K

ricki-z, 2021-02-06:
The CSV option will send the measured values as comma separated values to the serial interface. So you need to connect the NodeMCU directly to a computer. And you need to set the debug level to zero, otherwise there will be debug messages between the data lines.

ricki-z, 2021-04-02:
@Kreid4 can we close this issue?

Kreid4, 2021-04-02:
Yes thank you.
Regards
K

closed 2021-04-02 by ricki-z