opendata-stuttgart/sensors-software #873: Time not synced / NTP not working

RoTahnWaurie, 2021-02-09:
Due to some "vacation" days during the pandemic I have spare time to look at some issues I have noticed the last few days/weeks. It seems the sensor does not sync its clock to any ntp server. It always says `1970` and does not send data to madavi. NTP works perfectly fine for other devices in the subnet. No firewall is blocking traffic. Any suggestions how to fix this?

[image]

**Debug Level:**
```
airRohr: NRZ-2020-133/DE
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to <SSID>
.............
WiFi connected, IP is: <IPv4>
Starting Webserver... <IPv4>

ChipId:  <ID>

MAC Id: <MAC>
Read SDS...: 18-01-18(e50c)
Stopping SDS011...
Read DHT...
Send to :
sensor.community
Madavi.de
----
Auto-Update active...
validate request auth...
ws: root ...
ws: status ...
ws: status ...
ws: status ...
ws: status ...
ws: status ...
PM10:  13.37
PM2.5: 7.10
----
## Sending to sensor.community - SDS011
Time incorrect; Disabling CA verification.
Temperature (°C): -2.00
Humidity (%): 99.90
----
## Sending to sensor.community - DHT22
Time incorrect; Disabling CA verification.
## Sending to madavi.de: 
Time incorrect; Disabling CA verification.
Time for Sending (ms): 32914
PM10:  15.20
PM2.5: 8.30
```

ricki-z, 2021-02-09:
Could you please send a debug log that contains the firmware startup? Press the "reset" button for this while the debug log is shown.

RoTahnWaurie, 2021-02-10:
After some debugging in the dhcp of the wireless lan I found an issue with the default dns server of the Unitymedia Connect Box I am using. (All other devices in that subnet are using alternate dns servers.)
After restarting the wireless router the sensor works fine. 

I am currently thinking about the large gaps in my sensors data and if they occurred due to dns problems. Probably should throw out that Unitymedia crap...

closed 2021-02-10 by RoTahnWaurie

ricki-z, 2021-02-10:
If https was activated for the APIs Sensor.Community and Madavi.de then there were no transmissions as the certs couldn't be verified (validity date in the future).