opendata-stuttgart/sensors-software #875: SDS011 liefert keine Werte  NRZ-2020-133/DE (Nov 29 2020)

Beachbone, 2021-02-13:
Hallo zusammen,
auch wenn es das Thema schon das eine oder andere mal gab hat mir bis jetzt keins geholfen.
Habe einen Sensor entsprechend den Empfehlungen auf https://sensor.community/de/sensors/airrohr/
NodeMCU, SDS011, DHT22
Stand der Firmware: NRZ-2020-133/DE (Nov 29 2020)

Nun liefert der SDS011 seit einiger Zeit keine Daten mehr und ich hatte heute endlich mal Zeit mir das genauer anzuschauen.
Prinzipiell wird der SDS011 vom NodeMCU angesteuert. Der Lüfter läuft für die Messung, die Status LED blinkt mit ca 1Hz aber es werden keine Daten angezeigt. Nur die Bindestriche.
Getestet habe ich:
-Verschieden Netzgeräte
- Einen 2. SDS011
- Deaktivierung von PiHole
- Whitelisting von madavi.de und sensor.community als Wildcard im PiHole
- Deaktivierung der Übertragungen an sensor.community
-  Hard-Reset, Soft Reset, Neustart, längere Zeit ohne Spannung

Kennt jemand eine Software mit der man den Sensor direkt unter WIN10 mal testen kann? Per beigefügtem CH340?
Oder ist der SDS011 einfach so bescheiden das er nur eine sehr kurze Lebensdauer hat?

Ich sage mal Danke im Voraus.

J.
[image]
[image]

RobTranquillo, 2021-02-17:
Hast Du schon mal auf dem webinterface unter `/debug` geschaut, ob da die Messung stattfindet und ob Werte gemessen werden? Wenn Du das Debuglevel auf Maximum stellst, schaut es dann zB. so aus:
```
R/ SDS011
/R SDS011
R/ SDS011
/R SDS011
R/ SDS011
/R SDS011
R/ SDS011
/R SDS011
R/ SDS011
PM10 (sec.) : 17.30
PM2.5 (sec.): 11.20
```

Dannach werden die Werte an die eingestellten Dienste gesendet:
```
----
## Sending to sensor.community - SDS011
Succeeded - api.sensor.community
R/ DHT22
Temperature (<0xc2><0xb0>C): 5.90
Humidity (%): 99.90
----
/R DHT22
## Sending to sensor.community - DHT22
Succeeded - api.sensor.community
```

ricki-z, 2021-02-17:
Es gibt für Windows eine Software, um den SDS011 über den meist mitgelieferten CH340 auszulesen:
http://inovafitness.com/en/a/chanpinzhongxin/95.html (unter dem Reiter "Downloads")
Die ist sehr einfach gehalten und u.U. zunächst mit chinesischen Texten. Das sollte sich zumindest auf Englisch umstellen lassen. Und zumindest zum Testen sollte es funktionieren.

Beachbone, 2021-02-27:
Bin nun wieder zuhause und konnte mir das genauer anschauen.
Die Software hatte ich schonmal getestet hat aber den COM Port nicht gefunden.
-> Muß im Kompatibilitätsmodus für WIN7 gestartet werden. Dann klappts auch damit.

Den Hauptfehler habe ich nun auch gefunden. Eine gebrochene Lötstelle.
Werde nachher meinen Lötkolben suchen und das reparieren. Hoffe dann geht alles wieder.