opendata-stuttgart/sensors-software #878: Add support for CO2 sensor Senseair S8

zefanja, 2021-02-22:
I'm living in South East Asia where we have air pollution almost all year round. We are using the luftdaten sensors in our school to measure PM2.5. levels indoors and outdoors. In addition to the PM2.5 sensors, we like to monitor CO2 indoors to check the CO2 levels during classes. 

I'm thinking of a Senseair S8 sensor, that you can use indoors / outdoors (-40°C to 85°C). It is a maintenance-free sensor and levels itself.

More details (including datasheet): https://senseair.com/products/size-counts/s8-residential/

If there is anything I can do, to support the integration if this sensor I'm willing to help.

pjgueno, 2021-02-22:
Hi,
Thanks for your message and your support.
I don't know if it is really interessant to meaure the CO2 level of closed private places for an open source project about environmental air quality data...
Yu can also built you own sensor which send info to Sensor.Community like the normal sensor.
Concerning the addition of new sensors, we are facing a problem with the size of the firware which should allow autoupdate. That means: the size of the firmware can't be bigger than half the size of the flash memory.
Anyway you can develop in the beta branch on github and integrate the sensor in our code. I can't remember if a CO2 attribute was already integrated in the API/DB. 
I will ask this evening for the current policy of the project concerning the CO2.

ricki-z, 2021-02-22:
First comment:
Operating temperature: 0 .. 50°C
Storage temperature: -40..70°C (or ..85°C, depending on model)

Even with global warming we had temperatures below -10°C for some days :wink: So these sensors aren't really outdoor sensors.

ricki-z, 2021-02-22:
Second comment:
Please read the part about ABC (Auto Baseline Calibration) in the spec sheet. And then think about a room that you can't ventilate completely because of very low temperatures outside. The CO2 concentration will be above 400ppm at every time, let's say 450ppm. After 8 days (default) these 450ppm will become the new 400ppm. People will stop ventilating earlier, concentration will be above the "new" 400ppm the whole time. This higher values will become the next new 400ppm and so on ...
So as long a you can't guarantee to have the lowest possible CO2 level once every 8 days you will have a problem with comparing values over a longer time.

zefanja, 2021-02-22:
> Operating temperature: 0 .. 50°C
> Storage temperature: -40..70°C (or ..85°C, depending on model)

Right...my mistake.

> Even with global warming we had temperatures below -10°C for some days wink So these sensors aren't really outdoor sensors.

That is not an issue, because we only want to measure CO2 inside and not outside - and even if we wanted to - in the tropics you don't have -10°C :)

> Please read the part about ABC (Auto Baseline Calibration) in the spec sheet. And then think about a room that you can't ventilate completely because of very low temperatures outside. The CO2 [...]
So as long a you can't guarantee to have the lowest possible CO2 level once every 8 days you will have a problem with comparing values over a longer time.

That's right. Right now we have some other CO2 sensors in a few classrooms and it normally takes only 8-12h to get the CO2 levels back to the outside levels inside a classroom. The reason is, that here in SE Asia the houses and rooms are not that airtight compared to houses in Germany. There are always slots under the doors, the windows does not close very tight and so on. So this won't be an issue here.

zefanja, 2021-02-23:
Here is some code of the other sensor we are currently using: https://github.com/airgradienthq/arduino/blob/0753a4c9dd57d2d23f978b34b1cf7f787ea52571/AirGradient.cpp#L567

CobraCarGuy, 2025-04-24:
For applications where the environment is occupied 24/7 like a hospital, Senseair recommends setting the ABC Period to 3 to 6 months. The default period is 7.5 days (180 hours). Having a long ABC period increases the chance that a measurement will be taken when outdoor levels of CO2 are present. Sensor drift over 3 to 6 months is minimal with our sensors. All of these have a great deal of programmability. Zero calibrations are best, but those must be done manually and require nitrogen or argon.
Our newest sensors (Sunrise/Sunlight) have an adjustable ABC Target, so if the levels are known, they can be set here in order to dial ABC in.

CobraCarGuy, 2025-04-24:
@zefanja If the classrooms do get down to outdoor levels of CO2 within 8 to 12 hours as you mentioned, then ABC should work great for you.