opendata-stuttgart/sensors-software #879: BMP280 is shown in configuration file instead of BME280

boge717819, 2021-03-09:
hello,
my board has a BME280 attached to it, but this is not shown in the tab with "values". Instead the BMP280 is shown with empty values (see screenshot). The latest firmware is used. This happens on different boards and with different BME280's. What am I doing wrong?

thanks in advance,

Etienne

[image]

pjgueno, 2021-03-09:
I had and I have again exactly the same problem: the solderings oydize. Use some Kontakt 60 desoxydizer or some 95% ethanol with a toothbrush to clean gently the pins and solder points. It should start again and be recognized as a BME280.

boge717819, 2021-03-10:
Hi Pierre-Jean,

this morning I tried your solution and it worked!  I had to resolder the
connections on the BME280. They also were heavily corroded. We are back in
business, thank you very much.

Danke, thanks, merci

Etienne

ropeters68, 2021-03-10:
You can use a little nail polish to protect the connections.

boge717819, 2021-03-10:
indeed, nail polish could prevent this. Thanks!

Etienne

closed 2021-03-11 by pjgueno