opendata-stuttgart/sensors-software #881: Sds011 Bme280 keine automatische Anmeldung im Wlan

andreas735, 2021-03-19:
Hallo zusammen. Meine Messung läuft perfekt. Wird über Solar und Powerbank betrieben. Nachts ist dann irgendwann die Batterie leer. Am kommenden Morgen steht zwar wieder Spannung zur Verfügung,  jedoch erfolgt keine Anmeldung am WLAN Router. Erst wenn ich den Sensor Spannungsfrei schalte, erfolgte kurz nach dem Zuschalten die automatische Anmeldung.  Kann mir hier jemand weiterhelfen? VG

pjgueno, 2021-03-19:
Ich verstehe nicht in wie fern es mit dem Sensor-Firmware zu tun hat. 

Zudem : Wo in den Anweisungen auf der Internetseite werden "Solar und Powerbank" erwähnt?

Bitte auch auf Englisch bei Github fragen stellen.

Bitte die Stromstärke der Powerbank prüfen. Die Spannung kann genugend sein ohne es 1000 oder 1500 mA abgegeben werden, um das ganze System einzuspeisen. Das heisst: der Sensor schaltet sich ein, sobald es die 5V kriegt aber es schaltet sich gleich aus, weil es nicht genug Strom gibt. Und so, und so, und so... unendlich. Konsequenz: die Powerbank entlädt sich stets und das Sensor fängt nie an Daten zu messen.

Deswegen empfehlen wir ein USB-Charger...

Sir können auch das Forum von Sensor.Community benutzen (siehe Internetseite).

RikDrabs, 2021-03-19:
Sorry to answer in English. Am i right that your supply voltage at night drops below operating range of the ESP?
Then this seems to me like the rising edge of the supply voltage in the morning is not steep enough to trigger a good reset signal, at a time when the voltage on the ESP8266 is again within operating range the next morning.
What you need is a circuit to interrupt the supply voltage to the meter (ESP8266+SDS011+DHT22/BME280+rest of the circuit), and only re-establish that supply voltage when it is in range (4.8 Volts to 5.2 Volts), and stable enough to supply starting currents (1 Amp). A bit like you're doing now manually. Electronic circuits for doing so exist. Look on the internet, if you don't have the electronic experience yourself.

andreas735, 2021-03-21:
Thanks a lot for the answers.
In my opinion it must have another background. As soon i disconnect and connect the Power supply in the next early morning the system runs immediately.

pjgueno, 2021-03-21:
Yes, that's exactly what we told ! if you disconnect, the powerbank can reach the 1500 mA. You have to build a kind of system which can wait for a certain load.

andreas735, 2021-03-21:
Thank you. I will try this.

closed 2021-03-21 by andreas735