opendata-stuttgart/sensors-software #890: Adding support for neopixel ring

rotor-studio, 2021-04-09:
Hello dear community. 
I m looking for integrate a more intuitive way to read the sensor values for educational project. My idea is to connect a neopixel ring to see directly de values by colors or drawings, adding the capabilites of play with data directly over the sensor. Simi lar like this:

Led ring: https://www.adafruit.com/product/1586
An example of playing with this rings: https://www.romantorre.net/v4/wp-content/uploads/2020/12/IMG_7877-1024x682-1.jpg

I was wondering what would be the best way to integrate the codeon a custom firmware. And obiusly, i can write arduino or python code but if someone can oriented me the most directly way, i can contributte with this. 

someone can  help me?...Thanks!!

pjgueno, 2021-04-09:
Firstly, you should read and understand the whole firmware : https://github.com/opendata-stuttgart/sensors-software/tree/master/airrohr-firmware... 
You can use these variables:
pm10_serial
pm25_serial

You have to add a library .

https://www.youtube.com/watch?v=6U4JssgWjr8
http://www.esp8266learning.com/some-neopixel-examples.php

I am not sure there is still a pin to plug the NEopixel...

bertrik, 2021-04-10:
Hi, I've build a little light that does approximately what you describe I think.

It's an ESP8266 Wemos D1 mini board with a LED shield that connects over WiFi, does a wifi-AP-based geolocation to determine its approximate position, queries the sensor.community API to find the closest particulate matter measurement station. Then it displays the particulate matter concentration as a color, updated every 5 minutes, from deep blue through yellow/red to bright purple.

You can find the project page here:
https://revspace.nl/StofAnanas

Ive been toying with the ESP-now protocol, to make it completely zero-configuration. ESP-now is a kind of peer-to-peer connectionless WiFi protocol, running at a lower bitrate (1 mbps) so with a slightly longer range than regular WIFi, supported in both the ESP8266 and ESP32. So the idea is that you just put in range of the particulate matter sensor and they will auto-negotiate even without needing an access-point. Data is transmitted as a JSON in a short packet. So it can control a simple RGB lamp, but also you can show the values on a small display.

Prototype code is here: https://github.com/bertrik/pmlamp-espnow-proto

My plan was to integrate this into a prototype of the airrohr-firmware, but I haven't gotten to that.

rotor-studio, 2021-04-12:
> 
> 
> Firstly, you should read and understand the whole firmware : https://github.com/opendata-stuttgart/sensors-software/tree/master/airrohr-firmware...
> You can use these variables:
> pm10_serial
> pm25_serial
> 
> You have to add a library .
> 
> https://www.youtube.com/watch?v=6U4JssgWjr8
> http://www.esp8266learning.com/some-neopixel-examples.php
> 
> I am not sure there is still a pin to plug the NEopixel...
> 
> Actually it would be a lot easier to call the value of a very sensor with our API and display the color every 5 minutes or so
> [https://data.sensor.community/airrohr/v1/sensor/{apiID}/](https://data.sensor.community/airrohr/v1/sensor/%7BapiID%7D/)

Thanks a lot. I was checking a little bit al the file tree and i can understand better now...I will try to compile it in a custom versión.  And yeah, there are no voltage pins, i will bridge it for getting 5v. 
The call seems usefull also for future projects, thanks!!!!

rotor-studio, 2021-04-12:
> Hi, I've build a little light that does approximately what you describe I think.
>
> It's an ESP8266 Wemos D1 mini board with a LED shield that connects over WiFi, does a wifi-AP-based geolocation to [...]

Thanks a lot, is basically the same idea but with another desing totally based on airrohr firmware. Congrats. 
I will try to add neopixel ring adding the libaries to the main code and i will share my code also. 
Best!