opendata-stuttgart/sensors-software #891: Falsche Uhrzeit

compidev, 2021-04-10:
Mein Sensor schreibt mit folgendem Script jeden Tag CSV Dateien auf meinen Pi: https://github.com/opendata-stuttgart/madavi-api/blob/master/data_simple.php

Die Uhrzeit die angegeben wird weicht aber von der richtigen Zeit um 2 Stunden ab.

Das Steht bei mir im Gerätestatus:
`

NTP Info | 45.33.84.208 (0.pool.ntp.org) reachable: Yes
-- | --
Zeit (UTC) | Sat Apr 10 17:44:55 2021

`

Kann man die Zeit irgendwie per Einstellungen ändern? Wenn nicht, würde ich mir eine Funktion wünschen, mit der man eigene Zeitserver und Zeitzonen festlegen kann :)

pjgueno, 2021-04-10:
> Die Uhrzeit die angegeben wird weicht aber von der richtigen Zeit um 2 Stunden ab.

Die richtige ist die UTC, die vom LNE in Paris festgestellt ist: [https://syrte.obspm.fr/cgi-bin/heure_op_js](https://syrte.obspm.fr/cgi-bin/heure_op_js)

Look here: 
[https://stackoverflow.com/questions/5806526/php-utc-to-local-time](https://stackoverflow.com/questions/5806526/php-utc-to-local-time) 
[https://stackoverflow.com/questions/3792066/convert-utc-dates-to-local-time-in-php](https://stackoverflow.com/questions/3792066/convert-utc-dates-to-local-time-in-php)

compidev, 2021-04-13:
Aber eine Funktion zum Setzen der gesendeten Zeit wär trotzdem sinnvoll oder?

ricki-z, 2021-04-14:
Leider ist nicht immer sichergestellt, daß die Uhrzeit auf den NodeMCUs korrekt ist (Fritz!Boxen haben z.B. für Gastnetze die Abfrage der NTP-Server gesperrt). Daher sollte die Zeit besser auf dem genutzt werden, der die Daten empfängt.
UTC ist einfach die 'stabilere' Lösung gegeüber der lokalen Zeit, da hier nicht durch Wechsel zwischen Sommer- und Winterzeit Überschneidungen oder Lücken entstehen.
Die meisten Programmiersprachen sollten Funktionen zur Umwandlung der lokalen Zeit nach UTC mitbringen.

compidev, 2021-04-14:
Ok, macht sinn!
wäre es für einen von euch möglich das Script so zu ändern dass es 2 Stunden an die gesendete Zeit ranhängt?

ricki-z, 2021-04-15:
Das Script sollte eigentlich UTC ausgeben. Von daher sollte eventuell geprüft werden, welche Zeit für welche Zeitzone eingestellt ist. Ist die aktuell eigestellte Zeitzone eventuell schon UTC, aber die Zeit ist die lokale Zeit?

closed 2023-06-22 by compicatt