opendata-stuttgart/sensors-software #896: Add 802.1X support (ex. Eduroam and authenticated network sessions)

vasconeves, 2021-05-11:
Hi!

I'm using about 50 feinstaub sensors but I have this recurring problem: school and universities have authenticated networks that require adding more information than just the network name and password. So I need to get creative and use a lot of workaround which are not really ideal in terms of security.

I'm wondering if this feature can be added in future updates.

Thanks.

Best regards,

Vasco

sidietz, 2021-06-10:
Moving to ESP32 may help because it offers more SDRAM, a low-power second core and more computing power.

That does not change the fact, that lots of sensors-software code would have to be refactored/re-written, then.

vasconeves, 2021-06-10:
Hey,

Maybe we should start thinking about making branches for different
builds/cores for the auto update. This is a huge problem for penetration in
universities and educational networks, at least in Portugal.

Nevertheless we have now much more sensors as before, up to 30 and counting
:)

Usually I suggest to place a router (or any routing equipment) from inside
the network but this poses considerable danger in terms of security.

Cheers,

V.