opendata-stuttgart/sensors-software #897: BME 280 does not transfer data.

Truffaldino13, 2021-05-13:
Hello.
I have a problem.
BME 280 does not transfer data to https://maps.sensor.community/#2/0.0/0.0 
Before that, there was a BMP 280 sensor and there were the same problems. 
All values are on the device address. When trying to transfer data from the BME 280 sensor, an error occurred.
Debug info: 
validate request auth...
ws: debug level ...

/R SDS011
PM10:  2.25
PM2.5: 1.63
----
## Sending to sensor.community - SDS011
Succeeded - api.sensor.community
R/ BME280
Temperature (°C): 21.27
Pressure (hPa): 99274.09
Humidity (%): 40.32
----
/R BME280
## Sending to sensor.community - BME280
Request failedR/ SDS011

/R SDS011
PM10:  2.45
PM2.5: 1.90
----
## Sending to sensor.community - SDS011
Succeeded - api.sensor.community
R/ BME280
Temperature (°C): 21.24
Pressure (hPa): 99282.59
Humidity (%): 40.28
----
/R BME280
## S
Please help, I would like the device to have a finished look.
Device Information: 
ID: 4648114 (9c9c1f46ecb2)
Прошивка: NRZ-2020-133/RU (Nov 29 2020)

pjgueno, 2021-05-14:
It works however : [https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-4648114](https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?orgId=1&var-chipID=esp8266-4648114)

What about the configuration on devices.sensor.community? Did you choose the right sensors? What is the ID on the map ? Or just give me the geocoordinates of the location.

ricki-z, 2021-05-14:
@Truffaldino13 I have changed the sensor type from DHT22 to BME280 on devices.sensor.community. Now you should see the sensor on the map.

Truffaldino13, 2021-05-14:
Все в порядке. Спасибо.