opendata-stuttgart/sensors-software #902: Language change does not work

netmaniac, 2021-06-17:
New nodeMCUv3, flashed using flasher tool with latest_en.bin. After WiFi configuration it reboots, connect to WiFi. Then I try to change language on config page.  NodeMCU reboots, but still interface language is EN. In config dropdown shows PL as current  language (so config was written). 

During startup on serial:

```
Connecting to [CUT]
....SNTP synced: Thu Jun 17 09:56:30 2021

twoStageOTAUpdate
R/ SDS011 version date
/R SDS011 version date
HTTP GET: firmware.sensor.community:443/airrohr/update/latest_pl.bin.md5
GET r: -1
.
WiFi connected, IP is: 192.168.1.164
Starting Webserver... 192.168.1.164
```

ricki-z, 2021-06-17:
It looks like the firmware couldn't download the md5 file. So first you should try another reboot (if it was only a temporary server problem). If this doesn't work please check if the device is allowed to download files from external servers via https.

netmaniac, 2021-06-22:
I m away for couple days. After I return I do test language change.

ricki-z, 2021-06-23:
There were some ciphers for HTTPS missing on the server. So the firmware couldn't connect and download. The missing ciphers were added and the language change should work again.