opendata-stuttgart/sensors-software #903: Continuous restarts

jboev, 2021-06-29:
Hi,
Recently the sensor couldn't be read in local network as it is restarting infinitely.
The sensors map shows the sensor outputs without problems and I can see the records on Grafana also.
When the counter of seconds to first measurment expire it looks that connection with sensor is lost for few seconds, and when connection is restored the counter starts again. I restarted the sensor few times and router also but the issue still remains.
I've noticed that issue after a big local storm, when i tryed to check the outside temperature readings.

Output from debug is:  
ChipId: 6509912

MAC Id: 8caab5635558
Read SDS...: 18-11-16(cbc0)
Stopping SDS011...
Read BMxE280...
Trying BMx280 sensor on 76 ... found
Send to :
sensor.community
Feinstaub-App
MadaviairRohr: NRZ-2020-133/BG
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to VIVACOM_B
....SNTP synced: Tue Jun 29 07:49:03 2021

twoStageOTAUpdate
output debug text to displays...
.
WiFi connected, IP is: 192.168.1.25
Starting Webserver... 192.168.1.25

pjgueno, 2021-06-29:
[image]

https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-6509912

Seems to work quite good anyway.

It can be an issue with the USB charger. Check if there are at least 2000 mA in the specs otherwise when the SDS011 starts it sucks too much current and the boards stops...

jboev, 2021-06-29:
For me there is a problem as the sensors continue with restarts even after reflash with new firmware.
The USB charger used is 1.55A, but i do not assume that to be a reason because last few months everything was ok.
To exclude the PS as a reason i will supply the ESP8266 board directly from to pins with powerfull PS and will report the result. 
The reason for restarts from Sensor Status display is: 
Reset Reason | Software Watchdog

Petarkir2000, 2021-06-29:
Hello Jeko @jboev ,
I would prefer the followed:
1. As above @pjgueno said please change the power supply and check again. Please note that the chinеse power supply can change its parameter due to time or electrical shock. Additionally please note that 1.55A of power supply is not enough especially during the start of SDS controller.
2. Flash again the firmware with the tool, but before that erase the controller(It is a option in the flash tool). Some times when the power supply is not stable( you said that it was a storm) wrong data can be saved in config part of controller too.

jboev, 2021-06-30:
Hi, 
I changed the PS with bigger one (no name, no power rated, but with 4 USB outputs - used to charge batteries for drones) and now everything working flawlessly.
It is a bit strange for me as the old one was Samsung 1.55A rated and it worked from the initial sensor setup without any problems (until last week). Actually the sensor sends the data to servers, but after that due to SW watchdog restarts - is it possible this issue to be due to the latest Firmware update?
Thank you all for the support on this issue!

Petarkir2000, 2021-06-30:
@jboev No, the issue is tipical for bad power supplies...

AirBG.Info

pjgueno, 2021-07-27:
>  changed the PS with bigger one (no name, no power rated, but with 4 USB outputs - used to charge batteries for drones) and now everything working flawlessly.

Gotcha!

closed 2021-07-27 by pjgueno