opendata-stuttgart/sensors-software #904: Sensor not collecting and transmitting data

aaaprahamian, 2021-06-29:
Hi, I have a sensor ID 2476676. Since 28th it stopped collecting and transmitting data. Firmware version: NRZ-2020-133/EN (Nov 29 2020). Can you please advise how can I troubleshoot the sensor and understand what the issue is and how to fix it? I have not changed anything and the stop is sudden.

pjgueno, 2021-06-29:
It sends data actually:
[image]
Can I have you location and/or the ID on the map ?

aaaprahamian, 2021-06-29:
Hi, thanks for the quick reply. The on the sensor community web page is: 55382. I am also using the aqi.eco but there is no data for sure. Do you think it could be an issue with the aqi? Also when I open directly the IP address of the sensor and go to the current data tan I see no data :(

aaaprahamian, 2021-06-29:
I am located in Sofia, Bulgaria if this would help.

aaaprahamian, 2021-06-29:
[image]

pjgueno, 2021-06-29:
Incredible ! Both the links are showing data:
https://maps.sensor.community/grafana/d/000000004/single-sensor-view-for-map?orgId=1&var-node=55382
https://api-rrd.madavi.de/grafana/d/GUaL5aZMz/pm-sensors?orgId=1&var-chipID=esp8266-2476676
It is a mystery...
Does the hexagon appears on the map at your location? Zoom and send me the url with the hash.

On your last picture, it is normal: you restarted and the time for the first measurement was not reached yet.

aaaprahamian, 2021-06-29:
Hi, yes the hexagon is still there. Here is the link: https://maps.sensor.community/#16/42.6552/23.2593

As for the data on current data no matter how much time passes there is no data. Also I see that after some time of stand by I am not able to connect to the IP of the sensor and I need to restart it.

Petarkir2000, 2021-06-29:
@aaaprahamian could you contact hq@airbg.info to find solution?

aaaprahamian, 2021-06-29:
Will do this now. 👍

closed 2021-06-30 by aaaprahamian