opendata-stuttgart/sensors-software #905: AirRohr - Connecting to corporate network - finding the MAC address

mjrVdL, 2021-07-06:
Dear all,

After searching the FAQ and the existing posts, I finally ask my questions here, hoping that it is the right place. My apologies if it's not, I'm new to this and never sure where to post.

I'm trying to setup an AirRohr device for my office, using my company's Wi-Fi, and it seems I ran into two problems. 

1. I am asked to provide the microcontroller's MAC address to identify it, but never have been able to find it for sure (looking for 6 hex numbers separated by semicolons). Where or how can I find it (found [this](https://github.com/opendata-stuttgart/sensors-software/issues/49) but it doesn’t really help)?

1. As the device should be identified by its MAC address, there is no password for the Wi-Fi. When I supply the SSID and leave the password field empty, it stays in configuration mode and doesn’t try to connect to the network (we thought we could spot the correct MAC address in the logs, but none showed up). Is there a way for it to work without password?

Thanks for your help.

pjgueno, 2021-07-09:
Before flashing the firmware, install this code with the Arduino IDE in order to get the MAC address of your ESP8266 board:
[https://techtutorialsx.com/2017/04/09/esp8266-get-mac-address/](https://techtutorialsx.com/2017/04/09/esp8266-get-mac-address/)
It basically return HEX. Copy the result from the serial monitor of the Arduino IDE and configure your company network.
The login without password should work when the network has the MAC Address. I used to connect to a network of mine without PW.

Phaze-III, 2021-07-10:
The MAC address is also shown in the header of the configuration page which shows 'ID: <digits> (hex-string)'. The hex string is the MAC address without the colons.

mjrVdL, 2021-07-10:
Thanks! 
@Phaze-III  I thaught it might be the MAC address, but since it didn’t show up in the router’s log, it seemed wrong. I’ll try again with help of our network admin. 
@pjgueno  I’ll still try to get it via the IDE, just for the fun of learning how to do it. 
Thanks again for your help.

closed 2021-07-10 by mjrVdL