opendata-stuttgart/sensors-software #909: connection failed

Giotastic, 2021-07-27:
Hi to everyone, I am Giovanni and I am not a computer genius so I really need your help please with the connection failed error

[image]
[image]
[image]
[image]
[image]

THANKS

pjgueno, 2021-07-27:
Hi,
I assume you read all the instructions precisely... You should have registered your sensor on devices.sensor.community with the correct ChipID, shouldn't you ?

No I am joking: your sensor is not registered.

> Node not found in the database

Do you really understand what all the checkboxes you have checked do?

Do you really have an account on all this instances?

- Feinstaub-App
- AirCMS
- Opensensmap
- a personnal API
- InfluxDB

It is a lot of administration for someone who 

> is not a computer genius...

You sensor seems to work anyway so please read this attentively: [https://sensor.community/en/sensors/airrohr#Register](https://sensor.community/en/sensors/airrohr#Register)

And I always tell: if it is your first sensor do exactly what is written in the guides.

If it were your first time in an helicopter, you would not press every buttons and push every sticks, don't you?

Giotastic, 2021-07-27:
thanks  
pjgueno I didn't touch any button at first and I even register my device and received weekly email saying my device wasn't sending data, but after many days I got fed up and started touching buttons (if the goal of  a project is to include everyone it should be  easier and straight forward) :)
I did register my  sensor but I don't know if I did it correctly. 
Is there anyone that can help me in the community to  understand what I did wrong please? thanks

pjgueno, 2021-07-27:
You registered something which is not sending data and your sensor which sends data is not registered...
The solution is easy: you haven't typed the right number. Always triple check!

> if the goal of a project is to include everyone it should be easier and straight forward) :)

Actually I still haven't understand how some brands succeed to sell 1000 € smartphones which are imho much more difficult to configure than a sensor of ours.  The configuration can be difficult because we do not collect your wifi credentials...

I can help you but we need to communicate with Email (I have to register your sensor and transfer it to your account). Send me an address.

Giotastic, 2021-07-27:
pjgueno Merci for your help. I re register inputting the ID i Saw on one of the tab and by luck I see my sensor on the map now......
it still say connection failed should I uncheck some stuff now like  the one  you mention before?
Feinstaub-App
AirCMS
Opensensmap
a personnal API
InfluxDB

pjgueno, 2021-07-27:
Just keep this:
[image]
Uncheck the rest!

pjgueno, 2021-07-27:
If it appears on the map. It should not say "connection failed". Send me the serial logs. And your chipID. I will check

Giotastic, 2021-07-27:
I actually deleted the id I registered in order to re-register and include hardware for humidity and temperature this time. So I went to register the the sensor and I get the following message 

This sensor ID is already registered.

but I have removed 10 minutes ago!

Is there a way o communicate with you privately without showing to everyone my email address but only to you? Merci

pjgueno, 2021-07-27:
Aie aie aie...
That's normal, you should not have deleted the sensor without asking me.
Send a message to tech@sensor.community with the Chip ID and ask for complete delete...
I don't want to publish my address here as well. 
Come to forum.sensor.community you will find me as pjg.

rotor-studio, 2021-07-27:
Hello!! 
I don´t know if is the same issue but some time a go, two of my sensors cant connect and i wonder me if there are any problem with the database. I don't know exactly the moment, two has disappearing
Thanks!! ;) 
[image]
[image]

pjgueno, 2021-07-27:
Come to the forum to be able to DM me: forum.sensor.community

rotor-studio, 2021-07-27:
> Come to the forum to be able to DM me: forum.sensor.community

Ok!! Thanks

closed 2024-04-03 by ricki-z