opendata-stuttgart/sensors-software #910: Add support for IKEA Vindrktning

mrueg, 2021-07-28:
Ikea has published a product that is collecting data on air quality https://www.ikea.com/de/de/p/vindriktning-luftqualitaetssensor-70498242

@hypfer has published a repository on how to read data from this sensor with an ESP8266: https://github.com/Hypfer/esp8266-vindriktning-particle-sensor
https://hackaday.com/2021/07/24/esp8266-adds-wifi-logging-to-ikeas-air-quality-sensor/

The particle sensor used is a PM1006, which is based on LED and not on laser detection. http://www.jdscompany.co.kr/download.asp?gubun=07&filename=PM1006_LED_PARTICLE_SENSOR_MODULE_SPECIFICATIONS.pdf

If the quality of the sensor is acceptable, this might be a cheaper and quicker option to get more sensors in the field as people could easily buy this item from IKEA without waiting for several weeks to receive a shipment from aliexpress/ebay/etc.

bertrik, 2021-07-28:
My opinion: It's a very nice light, I like how a big company like IKEA has built a very affordable thing that visualizes something you cannot easily see otherwise and this way adds awareness for airborne particulate matter as a factor in air quality.

I've opened the device , read the sensor datasheet, wrote some code to decode the frames.
I just think it is probably not a very serious alternative to an SDS011 sensor or even in the same class of performance.

I suspect the maintainers will be hesitant to include it, the goal of the firmware is not to support each and every device out there, just a couple of the well-known and well-characterised ones and support those in a stable way.

I do like the idea of having a simple visualisation of particulate matter, and this IKEA thing is simple and stylish.
Perhaps instead of re-using the sensor inside, you can just reuse the light / box, e.g. put in electronics so it visualizes measurements of your outdoor sensor.community sensor!

pjgueno, 2021-07-28:
Indeed!
Isn't it a GP2Y1010AU0F? Like in the Canari of Communautique: [https://github.com/echofab-communautique/projet-canari](https://github.com/echofab-communautique/projet-canari). It seems to be a kind of analogue voltage sensor.

ricki-z, 2021-08-04:
I'm not sure if the ESP8266 instructions are really working. They are changing the voltage (and with that the speed) of the fan. This should lead to less air flowing through the sensor, reducing also the amount of particles that are counted.
And the IKEA sensor is an indoor sensor. So I don't think that it's very useful to add this to our project.

Hypfer, 2021-08-04:
> I'm not sure if the ESP8266 instructions are really working.
Frankly, this is getting quite annoying. Why would it be documented if it wasn't working?

Anyways, I do agree that this sensor isn't suited for this project as there seems to be no calibration whatsoever.
See this user report:
https://github.com/Hypfer/esp8266-vindriktning-particle-sensor/issues/16#issuecomment-891603624

I've also experienced similar behaviour with my four sensors. Both with 3.3v as well as with 5v.

It would make sense that there is no calibration, considering that Ikea never intended to hand out actual ug/m³ measurements. Even uncalibrated data is more than enough for a simple green/yellow/red scale.

ricki-z, 2021-08-04:
As I said I'm not sure. I don't have any IKEA sensor here to check this. So maybe the fan speed is high enough even at 3.3V as the limiting factors may be the air vents of the case.

mrueg, 2021-08-04:
Thanks everyone for the feedback! I can see now why it is not a suitable addition to the sensors.community project.