opendata-stuttgart/sensors-software #913: Excessive lines BME 280

Nealux, 2021-08-19:
Testing the beta software there are five entries for the BME 280 sensor.

Both pressure and temperature have two lines.
Please see picture below

[image]

Firmware: NRZ-2021-134-B4/SE (Aug 10 2021)
What could the problem be?

ricki-z, 2021-08-19:
These are missing translations.
The second pressure line should be pressure at sea level. The second temperature line should be the dew point.

Nealux, 2021-08-20:
Thanks a lot for a fast answer clarifying this to me.
Hopefully this translation error will be corrected in the future.

ricki-z, 2021-08-20:
@Nealux You may help us by adding the missing texts in the file https://github.com/opendata-stuttgart/sensors-software/blob/master/airrohr-firmware/intl_se.h
Or you write down the 3 missing strings here.

Nealux, 2021-08-20:
I do not want to risc anything by doing the change myself…
The missing translations are as follows:
Pressure at sea level = Lufttryck vid havsytan
Dewpoint = Daggpunkt
Am i correct if I think that the line for ”Dewpoint” should look something like this?
const char INTL_DEWPOINT[] PROGMEM = "Daggpunkt";

Ulf

Skickades från E-post<https://go.microsoft.com/fwlink/?LinkId=550986> för Windows

ricki-z, 2021-08-20:
I have added the 2 translations.
The code `const char INTL_DEWPOINT[] PROGMEM = "Daggpunkt";` would be correct.
Do you think you could also translate the 4 "static ip" lines and 'height above sea level'? If these are translated then I will upload a new beta version.

Nealux, 2021-08-22:
Static ip = Statisk ip
Height above sea level = Höjd över havet

Regards

Ulf

From: Rajko Zschiegner 
Sent: Friday, August 20, 2021 10:49 PM
To: opendata-stuttgart/sensors-software 
Cc: Nealux ; Mention 
Subject: Re: [opendata-stuttgart/sensors-software] Excessive lines BME 280 (#913)

I have added the 2 translations.
The code const char INTL_DEWPOINT[] PROGMEM = "Daggpunkt"; would be correct.
Do you think you could also translate the 4 "static ip" lines and 'height above sea level'? If these are translated then I will upload a new beta version.

Nealux, 2021-08-25:
Static ip = Statisk ip
Height above sea level = Höjd över havet

Skickades från E-post<https://go.microsoft.com/fwlink/?LinkId=550986> för Windows