opendata-stuttgart/sensors-software #915: Impossible to connect to sensor after update (+ fan from the PM sensor keeps spinning)

MagTun, 2021-08-23:
My sensor was automatically updated but after the update I couldn't connect to it via its IP. It could see that it was connected to my wifi access point and that the IP I used was correct. When I open this IP address in my browser, I got a "no connect" error message (or something like that). 

A simple switch off (for 30 sec or so) and on solve it but this happened already several times after updates. Also sometimes the fan from the PM sensor just keep spinning until the next restart. If it happens at the beginning of the night, it can take hours before I notice it and manually switch it off and on again. This is be very bad for the PM sensor.

sumerland, 2021-08-23:
Same here - both of my airrohrs lost connection at the same time (yesterday 9:48 CEST). One restarted after a reboot of the WIFI repeater, ran for a few hours and stopped again. I power-cycled it today, again it ran for a few minutes and stopped again. The other sensor did not react to the repeater restart. I power-cycled it today and settings were lost. I configured again but a few minutes later it's not responding anymore. Beta OTA is enabled on both.

MagTun, 2021-08-23:
It's funny, because, like you, my sensor has just stop again right now! Once again I can still see that it is connected to my wifi access point but I can't access the IP as in my browser.  This time restarting it didn't work, so I guess, I will have to manually flash it. 

I also have a second sensor and I had no problem with it: 
Last OTA	20 hours, 44 min, 59s 
Firmware version: NRZ-2020-133/EN (Nov 29 2020)

sumerland, 2021-08-23:
The second sensor seems to run the current stable relesase (29.11.2020). It's the latest beta or the attempt to do an OTA. I will check if I can get access to the one which is now on the current beta and see if disabling OTA helps.

sumerland, 2021-08-23:
Both sensors up and running (134-B3 and 134-B4), with OTA disabled. Will follow this thread and enable OTA as soon as the cause of the issue has been found.

MagTun, 2021-08-23:
This is weird. I just switched on my failing sensor after 3h of it being switched off, and it working again (just like in my first message). I haven't changed anything in the meantime, it was just off. I wasn'T sure it is was a beta and I can now confirm that it isn't (Load beta firmware is unchecked). 

Firmware version: NRZ-2020-133/EN (Nov 29 2020)
Last OTA	3 min, 32s

MagTun, 2021-08-23:
.... and the error is back : I can see that it is still connected  to my wifi repeater via Putty but I can't access it in my browser via its IP address, nothing changed on my side since my last comment, it was running perfectly

ricki-z, 2021-08-23:
Could you please check this with another power supply? Many of the connectivity problems are related to wall chargers that doesn't deliver enough power. The wall chargers aren't really made to work 24/7.

sumerland, 2021-08-23:
I can do a check tomorrow. However, my two sensors are connected to different power supplies. The first one shares a 220V -> USB supply with a NAM sensor. The other is connected to a solar panel / battery fed power supply. Both stopped working at exactly the same point in time. Both stop working immediately or shortly after being booted with OTA enabled. Both run w/o issues with one of the previous betas and OTA disabled. But as said in the beginning, I will enable OTA on the first sensor and power it with a different supply.

ricki-z, 2021-08-23:
There were no changes to the release version in the last months. The latest version NRZ-2020-133 was published Nov 29 2020. The message on the status page is only showing the time of the last check for an update (we have to change this text ...).

sumerland, 2021-08-24:
I was able to upgrade to Aug 22 beta, but...

Sensor 1 & 2 both stopped working two days ago. Sensor 1 was manually flashed to Aug22 (still not working), then flashed with an older release and ran over night with OTA disabled. Sensor 2 could be restarted, kept on the previous beta by disabling OTA and survived the night. I activated OTA on sensor 1 this morning and it updated smoothly. I activated OTA on sensor 2 and again it became unresponsive. I flashed Aug22 fw with no success. I then flashed an older release (sensor works) and activated OTA again. Now sensor 2 was able to update to Aug 22.

Summing up, both sensors were not able to OTA from Aug 10 (I think this was the last release date I saw in the page header) to Aug 22 directly. Both needed to be flashed with Aug 22 followed by an older release. Then OTA worked.

Petarkir2000, 2021-08-24:
Absolutely the same problems as @sumerland

sumerland, 2021-08-24:
And it's continuing... Sensor 2 I accidentally flashed with the english version. It was running on Aug22 (english) and all I did was changing to DE in the config page. Restarted & lost the sensor again. I will now repeat the steps (flash Aug22DE, then older release, then OTA to DE beta) and report.

ProBackup-nl, 2021-08-24:
Same here: 2 out of 3 sensors running beta firmware with auto updates enabled are dead since August 22, between 8:07 and 9:04 according to e-mail notification message date/time stamp. According to stats. the "serial SDS011" values are: 2021-08-24 11:27:39 and 2021-08-22 20:31:15. I can ping the esp, I can't access the web UI.

The third sensor running NL NRZ-2021-134-B4 (Beta) firmware is unaffected. SDS011 last update according to devices. = 24.08.2021, 11:37:13, according to stats. = 2021-08-22 20:59:54.

Power supply is identical for all 3 sensors: a Renkforce PoE to USB adapter.

sumerland, 2021-08-24:
It seems we are mixing bugs here in this thread. The original posting sounded as if the sensor was running the beta FW and the symptoms were the same as mine. Later it turned out that the sensor was running the (master) release. So it's most likely a different issue than the one I#m facing.

Now we have two additional users of the beta branch running into problems two days ago. I guess we need a new topic?

ricki-z, 2021-08-24:
To affected users:
As there are also many devices with the beta firmware running without any problems we need to cause of your problems.
So could you please post here the following informations?
- activated sensor components
- APIs you are sending to
- router model the device is connecting to
- any other information that may be helpful

sumerland, 2021-08-24:
Sensor 1: SDS011, SPS30, BME280, SHT35 sending to SC, madavi, local influxDB
Sensor 2: SHT35 sending to madavi, local influxDB
both connecting to TPlink RE550

FWIW: both are now running flawlessly on NRZ-2021-134-B4/DE (Aug 22 2021) - in my case it seems as if the OTA update was not able to finish properly.

ProBackup-nl, 2021-08-24:
All: running NL NRZ-2021-134-B4 (Beta), SDS011, BME280 to SC, madavi
serial SDS011, router model
Sensor 1 (dead): 2021-08-24 12:23:05, Peplink AP One to Fritz!Box 5530, ESP is power cycled several times.
Sensor 2 (dead): 2021-08-22 20:31:15, Compal CH7465LG-ZG (Ziggo Connect Box), no remote site access
Sensor 3 (ok): 2021-08-24 13:01:03, Fritz!Box 7560.

Petarkir2000, 2021-08-24:
1. SDS011; BME280 x 2 sensors 
2. Madavi 
3. Unifi network - AP AC LR/AP AC Pro 
4. Both sensors were showed as connected to network, but did not send/received information including no respond to IP. I received requests for password enter to devices. After pre-flash with the stable ver. and OTA update with BG beta, power circles they start working. I suggest that first OTA update did not finished correctly.
Additionally I found increasing of WiFi samples/second - from 20 to 45

MagTun, 2021-08-24:
> I guess we need a new topic?

@sumerland, don't switch the topic, all the info are here, so it's okay, I will update the title/and add a message in my original post to focus on that. (After a full night with my sensor switched off I have switched it on again and haven't experience any issue today... I don't understand what happened but will open a separate issue if it comes back.) So you keep the topic as your issue is urgent.

Phaze-III, 2021-08-29:
I've also seen instability after an OTA upgrade from NRZ-2020-134-B3/DE (Apr 8 2021) to NRZ-2021-134-B4/DE (Aug 22 2021). After the upgrade the sensor kept resetting with "Software Watchdog" or "Exception" as reset reasons and was practically useless. After a power cycle (power off, a minute waiting, power on) the sensor kept and keeps running very stable.

A possibly related observation however is that OTA *downgrading* from NRZ-2021-134-B4/DE to release (removing the Beta-checkmark) renders the sensor unusable. Checking the serial output shows the following:
```
airRohr: NRZ-2021-134-B4/DE
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to <redacted>
......SNTP synced: Sat Aug 28 23:41:21 2021

twoStageOTAUpdate
R/ SDS011 version date
/R SDS011 version date
HTTP GET: firmware.sensor.community:443/airrohr/update/latest_de.bin.md5
GET r: 200
Update md5: f230be8f7db0e0929129c38cd3aa3fc9
Sketch md5: 9478002f481e0bcf7c9c13ee75a87016
HTTP GET: firmware.sensor.community:443/airrohr/update/latest_de.bin
GET r: 200
```
and then hangs without any activity at all. Power cycling doesn't help, only re-flashing.

Active sensors: BME280, SDS010
API: Sensor.Community (HTTPS), Madavi.de (HTTPS), Local InfluxDB
AP: Ziggo Connect Box (Compal CH7465LG-NCIP-6.15.30-1p4-NOSH)

mnin, 2021-09-29:
I have the same issue, for a few weeks my dust sensor has been inaccessible because probably the latest firmware was installed.

If I block it on wifi, then I can even open the web page for configuration.

But as soon as it can log back into the wifi, it gets an IP address and then no longer responds to any request.

The attempt to flash the last beta firmware has not succeeded, even there the firmware then stops with

[image]

I then flashed the last stable version from last year, which runs again since yesterday evening.