opendata-stuttgart/sensors-software #916: airrohr sensor accesspoint fordert Netzwerkschlüssel

Locutosz, 2021-08-23:
Hallo,
seit fast einem Jahr Betreibe ich einen Luftdatensensor. Heute habe ich nun eine Mail erhalten, dass der Sensor keine Daten mehr liefert. Ich habe das überprüft und festgestellt, dass der Sensor nicht mehr in mein Wlan eingeloggt war. Nachdem ich die Stromversorgung kurz unterbrochen habe, habe ich festgestellt das der Sensor wieder einen eigenen Accesspoint auf macht. Ich nehme an, das die Firmware(wahrscheinlich beta, hatte dieses aktiviert) aktualisiert wurde, da der Sensor jetzt statt Luftdatensensor jetzt airrohr heisst. Der Accesspoint der neuen Firmware verlangt jetzt von mir einen Netzwerkschlüssel. Dies war bei der alten Firmware nicht so. Ist der der Accespoint bei der neuen Firmware jetzt wirklich passwortgeschützt oder ist da bei Firmwarewechsel etwas schief gelaufen? Kann mir jemand sagen, wenn der Accesspoint wirklich Passwortgeschützt ist, wie dieses heißt? Würde mich freuen.

Mit freundlichen Grüßen
Locutosz

sumerland, 2021-08-23:
Könnte mit den in #915 geschilderten Problemen zusammenhänge. Wobei ich mich mit dem vom Sensor aufgemachten Accesspoint ohne Schlüssel verbinden konnte. Er hat sich dann nach einem Neustart direkt aufgehängt.

Locutosz, 2021-08-23:
Hallo,
vielen Dank für Ihren Hinweis. Da stell ich mal die Frage an die Programmierer, warum diese fehlerhaft programmierte Firmware, trotz Kenntnis des Fehlers, weiter verbreitet wird. Da hilft wohl nur das Flashen der nonbeta Firmware. Bedanke mich schon mal für die extra Arbeit. :(

ricki-z, 2021-08-23:
@Locutosz Es Sinn und Zweck einer Beta-Firmware, dort die noch vorhandenen Fehler zu finden. Nicht alles, was passieren kann, passiert auch auf den Systemen der Entwickler. Viele der Systeme, auf denen die Beta läuft, sind nach dem Update wieder online (scheinen also keine Probleme zu haben).
Aber wer es besser kann, kann die Fehler ja beseitigen helfen :wink:

ricki-z, 2021-08-23:
Zum Passwort für den Sensor-Hotspot: dieses lautet in der aktuellen Beta, wenn nicht anders gesetzt, 'airrohrcfg'
Die derzeitige Releaseversion öffnet noch einen offenen, passwortlosen Hotspot. Da einige Leute dies als Sicherheitsproblem sehen, wird es in Zukunft an dieser Stelle ein Passwort geben.

Locutosz, 2021-08-23:
@ricki-z mir ist schon klar, das eine beta auch schwerwiegende Fehler enthalten kann. ;)  Vielen Dank für das Passwort. Das Passwort wird tatsächlich angenommen und ich bekomme zum Sensor Verbindung. Leider wird mir bei Aufruf der Seite http://192.168.4.1/ der Zugriff verweigert. Muss ich eventuell bei der beta eine andere IP Adresse aufrufen?

ricki-z, 2021-08-23:
Die IP sollte passen. Da hat sich nichts geändert. Was genau ist die Fehlermeldung (evtl. Screenshot)?
Sollte ein Passwort für den Zugriff auf die Seite im normalen Betrieb (in der Konfiguration unter 'Basic Auth') gesetzt worden sein, müsste das hier auch funktionieren.

Locutosz, 2021-08-23:
Da kommt die Meldung: access denied
Ein Passwort für den Zugriff habe ich nicht gesetzt.

Locutosz, 2021-08-24:
Soooo..... ich habe jetzt das Teil noch einmal auseinander gebaut und neu mit der Stable Version geflasht. Was soll ich sagen, ich hatte eine Verbindung und kam auch nicht rauf. :( Nochmal nachgedacht und einen Ubuntu Livestick erstellt. Gebootet und es geschah ein Wunder. Ich konnte mich mit dem Sensor einwandfrei verbinden. :) Sieht so aus, als ob die Sicherheitseinstellungen von Win 10 2004 21H1 den Kontakt zum Sensor unterbinden. Der Sensor hat sich mit meinem Router verbunden und läuft jetzt wieder einwandfrei.

ricki-z, 2021-08-24:
Es ist sooo nervig ... Mit jedem 2. Upgrade gibt es neue Probleme. Dann werden wir uns mal auf die Suche machen, warum Win10 21H1 da rumzickt (auf meinem Win 10 ging es seltsamerweise).
Im übrigen auch gern problematisch sind Antivirenprogramme mit "Firewall-Komponente", die dann mal eben Seiten sperren.

Locutosz, 2021-08-24:
Da ich den Sensor noch offen vor mir liegt habe ich mir mal getraut die aktuelle beta NRZ-2021-234-B4/DE August 2021 wieder installieren zu lassen. Diesmal wurden die Zugangsdaten vom Router nicht gelöscht. Alles funktioniert. Ich kann auch über den Router auf die Oberfläche zugreifen. Scheint als wär das Zugriffsproblem nur, wenn der Sensor einen Accesspoint aufmacht.
Kann man denn irgendwo informationen über die Beta nachlesen?

ricki-z, 2021-08-24:
Infos zur Beta gibt es hauptsächlich im Source code im Beta Branch hier auf Github. In der Versions.md sind (fast) alle Neuerungen und größeren Änderungen zu finden. Über die Historie kann man dann die einzelnen Pushs ansehen. Die sind meist kommentiert, wenn es nicht nur um die Korrektur eines Schreibfehlers ging.