opendata-stuttgart/sensors-software #922: How to keep continuous data after changing the NodeMCU? (Feature request: an ID not linked to specific nodMCU)

MagTun, 2021-08-31:
For some reason, I need to change my nodeMCU. But when I do that the ID of my sensor changes and so, I lose the continuity in my data. By continuity I mean that the things like the graph on madavi.de/grafana and the CSV on madavi.de/csvfiles.php

So my question is : how can I can I keep this continuity ?

 If it's no currently possible, it would be great to have a feature that allow us to switch the nodeMCU without loosing the continuity in the data. The feature request is then: the data should not be attached to the hardware's ID, but to something more flexible (eg. a ID number linked with their data that the users can use with any nodeMCU)

RikDrabs, 2021-09-01:
This is a long standing problem, that needs a solution, since replacing the ESP is a relatively common practice in case of WiFi problems. See issue #719 of 12 May 2020. We are supporting some 1000 active sensors from Brussels and surroundings, and this problem was encountered for between 25 and 50 times, each causing trouble in interpretation of local data.

WiliTest, 2021-09-04:
When we create a new sensor, we should be able to get a new dataset ID via the configuration page (http://192.168.1.39/config ).

This won't be a major change because the current system "ESP-ID" and the new system "Dataset-ID" can work alongside.  The Dataset-ID can even use the ID of the first ESP used to generate the data. That way Grafana and the csvFiles.php don't have to be change, the sensor will simply overwrite its ID (the default "ESP-ID")  before sending the data.

If we want to change the ESP but to keep the data continuity, we just have to copy the Dataset-ID and enter it in a field, in the sensor configuration page: `overwrite the ESP-ID, and send the data to this dataset linked with of this ID`. 

(The user should not have to go to the https://devices.sensor.community/sensors website to generate her dataset ID because not all sensor are registered there. Eg. I don't register a mobile sensor I use because I don't want it to show on the map. Generating the dataset ID and overwriting should be done via the configuration page of the sensor.)

RikDrabs, 2021-09-04:
Although changing the chip ID on the sensor's ESP seems to be the solution - also for me - there are a number of drawbacks associated with this. When the ID of an existing and still active sensor is used instead of that of the faulty sensor, the data from both active sensors with identical IDs gets mixed up. And this can happen accidentally, or even on purpose. In both cases, the data of the entire project becomes unreliable, if this were possible on a large scale. This solution is therefore avoidable.

RikDrabs, 2021-09-05:
For clarity:

Not registering an ID in https://devices.sensor.community/sensors is a bad habit that can provoke misuse by other users, and data corruption. Therefore not registering should be avoided. If the sole purpose is not showing up on the map you can declare the sensor "inactive" on the configuration page there.)

The case of not registering for the purpose of not showing up on the map should be clarified in the documentation!

pjgueno, 2021-09-06:
> The case of not registering for the purpose of not showing up on the map should be clarified in the documentation!

You are right, but people should understand that is is useless to participate in a citizen science project if you want to keep the data for your self... When people would understand that SC is doing nothing with the data while they keep sending extremely sensitive data through their smartphones and social network accounts...

WiliTest, 2021-09-09:
@pjgueno you are mistaken, I don't  want my sensor to show on the map because 1)it doesn't reflect any specific location 2) and most of the time not even the normal condition of the location (eg. I sometime use it when I work in some remote dusty environments)

RikDrabs, 2021-09-09:
But even then, not registering is not a good method. You should register, and put the sensor in non-active mode, to prevent misuse by accident or on purpose by other people. This to prevent data being mixed up between sensors. The reason is simple: if you don't register your own sensor with your own chip-id, then someone else might do so, by accident or on purpose. And that's the start of the mixup. Your data might show up on the map, at the location given by that other person. And anyone interpreting the data will not know, and take bad conclusions.
Remember: by using a Luftdaten sensor, you're not on your own, the sensor is coupled to the network anyway, even when not registered. Please DO NOT endanger the data integrity of the project by not registering, please!!!

RikDrabs, 2021-09-09:
@WiliTest Not registering is bad method for not showing on the map. You should register, and put the sensor in non-active mode, to prevent misuse by other people. This is to prevent data being mixed up between sensors and locations. The reason is simple: if you don't register your own sensor with your own chip-id, then someone else might do so, by accident or on purpose. And that's the start of the mixup. Your data might show up on the map at the location given by that other person. And anyone interpreting the data of the whole project (1000's of sensors) will not know, and possibly take bad conclusions.
Remember: by using a Luftdaten sensor, you're **never** on your own, the sensor is coupled to the network anyway, even when not registered. Please **DO NOT** endanger the data integrity of the project by not registering, please!!!
If you don't know how, or don't want to register, please give your actual chip-id, so that this can be done the right way, to protect the Luftdaten project.

pjgueno, 2021-09-09:
@WiliTest 

> it doesn't reflect any specific location (mobile sensor) 

So ask a maker to make a mobile sensor for you. The project SC is basically for **sensors with a specific location**. If you don't accept the project as is, why do you want to participate ? It is quite easy to do a sensor with a data logger on an SD card.

RikDrabs, 2023-10-13:
@pjgueno 
Is this issue already taken into consideration?
The need for switching from one Chip-ID to another, in case of failure of the ESP8266, and the need to keep continuity in the data in that case, doesn't go away.

RikDrabs, 2023-10-13:
@pjgueno 
To iterate on answers given by @WiliTest not wanting to register his sensor, reading them back 2 years later.
Why not block the graphics display for sensors that are not registered?
To prevent misuse of the project by users not wanting to register in the first place?
And showing a message: "please first register to show graphics"