opendata-stuttgart/sensors-software #923: dust sensor when it rains

kamomil, 2021-08-31:
Hi,
I build that dust sensor according to these instructions: https://sensor.community/en/sensors/airrohr/
 In the instructions it is shown that the dust sensor and the ESP8266 should stick out from the pipe. But what happen in case of rain? then those parts are note protected

ricki-z, 2021-08-31:
Please read the complete instructions!
After connecting the 2 pipe parts the SDS011 and the ESP8266 should be in the upper part of the "&cap;". Only a very short part of thin pipe (going to the SDS011) should stick out. The BME280 shold be placed on this thin pipe inside the larger pipe but as near on the opening as possible.

kamomil, 2021-09-02:
oh, I was not even aware that two pipes are needed. I thought there is only one. 
So the pipes are not identical? I bought the parts separately (not as a kit).
The pipe I have is currently as listed in the instructions: 'DN 75 87°'.

ricki-z, 2021-09-03:
The 2 pipe parts are identical. So the second one is also a 'DN 75 87°'. One end of the pipes is slightly larger than the other so that you can plug in the smaller end of the other part.

ClemensGruber, 2022-01-11:
Have a loot at e.g.: https://www.byteyourlife.com/wp-content/uploads/2017/04/Feinstaub_Luftdaten_selber_messen_Sensor-1200x640.jpg

[image]