opendata-stuttgart/sensors-software #926: BMP280 - shows constant -140.12°C

lmai95, 2021-11-04:
Hello everybody,

i built my Airrohr last friday, i discoverd that my BMP280 was reporting a rather low tempratur of -140.12°C. The presure had an reasonable value. So i started investigation with a different ESP8266. First I noticed that my sensor is reachable under the alternative address 0x76. Secondely I  could reproduce the problem on this µC as well. I know this might not be the right place for this issue but since I don`t find any where else, I thought this project has a big likelyhood of not so experienced user to encounter this problem. I'm a hardware engineer so I will probably solve it on my own. This might be caused by a cheap knockoff parts used or a faulty bootstrapping of the senor pins on the breakoutboard.

ricki-z, 2021-11-14:
I'm not sure, but I remember some power problems could cause this.

Phaze-III, 2021-11-15:
I also have the occasional extremes with my BME280, in my case temperature of -140.71, pressure 57733.13 and humidity 100%. After two years running this started to increase (every few hours).

Replacing the jumper wires with new ones did improve things. Now I only see this once every few days on average.

In my case a software reset resulted in 'sensor not found' which prevented the outliers but of course disabled the temperature/pressure/humidity readings.

What actually helps in my case is a power cycle. The BME280 will then again be found and works for a while. The fact that a power cycle does seem to fix things in a way suggests that power might be a factor but first thing to check is contacts/jumper wires.