opendata-stuttgart/sensors-software #929: Switch from BMP280 to BME280

willimuller, 2021-11-13:
Hi all,

I have started my airrohr with the BMP280. Now I wanted to switch to the BME280  to add the humidity meassurement. Unfortunately I don't see the new meassurement in the results. 

I have already checked the settings in the setting page of the sensor (http://<ipaddress>/config). The entry "BME280 (Temperatur, rel. Luftfeuchte, Luftdruck), BMP280 (Temperatur, Luftdruck)" is checked. 

In the settings site of https://devices.sensor.community/sensors/<number>/settings the sensor I switched the senspr to BME280. Is there anything else I need to do?

Thanks for your help

Willi

ricki-z, 2021-11-14:
The firmware should detect automatically if it's a BME280 or a BMP280 (at each startup). So are you sure that the new one is a BME280?
Changing the entry at devices.sensor.community should be enough to show the values on the map.

willimuller, 2021-11-19:
Well, I ordered a BME and this one has 6 Pins and the BMP has 4 pins. So I assume it is a BME. I didn't see a print on it.

ricki-z, 2021-11-20:
This is not a safe feature:
BMP280 with 6 pins: https://www.reichelt.de/de/de/entwicklerboards-temperatur-und-drucksensor-bmp280-debo-bmp280-p266034.html?PROVID=2788&gclid=EAIaIQobChMIwuP7yqCn9AIV7IxoCR2pOQ77EAQYASABEgL3RPD_BwE&&r=1
BME280 with 4 pins: https://www.az-delivery.de/products/gy-bme280?variant=12217435619424&utm_source=google&utm_medium=cpc&utm_campaign=SSC&utm_term=&gclid=EAIaIQobChMIweOn86Cn9AIVkbt3Ch3arwDYEAQYASABEgKAf_D_BwE

SirFlour, 2022-10-14:
i just changed my BMP with the new BME and it worked directly after the restart. I use the 3.3v versions and they look really silimar. Only a small sensor is a bit bigger. So i guess you might just have again a BMP. @willimuller: Could you solve it?