opendata-stuttgart/sensors-software #935: I don't know how to find my T/H data at archive.sensor.community

SFotonium, 2021-12-10:
Could someone help me locate the data from the temperature and humidity in the archive.sensor.community repository as I don’t know the code for the BME280 sensor.

I know my SDS011 is 5253 and for the old temperature/humidity DHT22 sensor was 5254 but I don't know for the BME280.

My ID: 5012919

Thank you in advance

pjgueno, 2021-12-10:
Are you sure your sensor is currently functioning ?

[image]

[image]

It should be the same number 5254.
Have you changed the configuration in both the config web page and devices.sensor.community ?

ricki-z, 2021-12-11:
@SFotonium The DHT22 was registered in our database but not the BME280. I have changed the DHT22 entry to a BME280 so the data should now be shown for ID 5254.
It would be nice if you won't flood all channels with the same request so that most of our team is working on this same problem. All other requests are solved with that and won't be answered separately.

SFotonium, 2021-12-12:
@ricki-z sorry for my persistence but I didn't receive any reply via email and I was advised to use various means to contact. Thanks. However, I don't see any logs at the repository for the 7th and 8th of December with the 5254. I can see the measurement here https://api-rrd.madavi.de/grafana/d/q87EBfWGk/temperature-humidity-pressure?var-chipID=esp8266-5012919&var-type=BME280&var-query0=sensors&orgId=1 but I need the logs.