opendata-stuttgart/sensors-software #936: Wrong format of 'pressure value' of a BMP280 Sensor

jj-duisburg, 2021-12-11:
Debug level from my sensor:

airRohr: NRZ-2020-133/DE
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to XXXXXXXXXX
..................SNTP synced: Thu Dec  9 12:40:04 2021

twoStageOTAUpdate
output debug text to displays...
.
WiFi connected, IP is: 192.168.178.XXX
Starting Webserver... 192.168.178.XXX

ChipId: 4909223

MAC Id: 84f3eb4ae8a7
Read SDS...: 18-11-16(5532)
Stopping SDS011...
Read DHT...
Read BMxE280...
Trying BMx280 sensor on 76 ... found
Send to :
sePM10:  43.00
PM2.5: 20.23
----
## Sending to sensor.community - SDS011
Succeeded - api.sensor.community
Temperature (°C): 4.50
Humidity (%): 90.70
----
## Sending to sensor.community - DHT22
Succeeded - api.sensor.community
Pressure (hPa): 101814.06                             **<---------------**

************************************************************************************

Raw data for Pressure is: 101814.06 hPa
Display data on the Website is: 1018.14 hPa, that seams to be the correct value.

Is there any format change for that data in the firmware NRZ-2020-133/DE?

I use  iobroker and the Luftdaten.info-Adapter from Matthias Kleine, to get the data from my sensor. 
Everything is ok except the pressure value of my new BMP280 Sensor. 
As Matthias told me that the Adapter only transport 1:1 the data from the firmware to the datapoints of ioBroker, something goes wrong. The pressure value is 100 times to high. I have to change  one data of six . That is not normal isn`t it?

I think that is an error in the firmware NRZ-2020-133/DE

Thanx for helping me/us.

Josef Jeßnitz

pjgueno, 2021-12-11:
The pressure is usually schown in hPa (hectopascal) but you can also show it in Pascal, mmHg, millibar ... 
It is no error just a divider/multiplier. You can easily write an integration to get the data in the format you need.

ricki-z, 2021-12-12:
The raw data of the pressure sensors is in Pascal. And we try to send only raw data to minimize the possibility of conversion errors in our firmware.

jj-duisburg, 2021-12-12:
> The pressure is usually schown in hPa (hectopascal) but you can also show it in Pascal, mmHg, millibar ...
> It is no error just a divider/multiplier. You can easily write an integration to get the data in the format you need.

> The raw data of the pressure sensors is in Pascal. And we try to send only raw data to minimize the possibility of conversion errors in our firmware.

I got the following data from debug: Pressure (hPa): 101814.06 <-------------

It is pascal 101814.06 or hectopascal 1018.14 but not the return data from debug. I think it is a
label error isn‘ it?

ricki-z, 2021-12-13:
Yes, this is a typo in the debug message.