opendata-stuttgart/sensors-software #942: New DHT22 sensor does not come to live

MJLEVER00, 2022-01-06:
Hello,

my DHT22 sensor has been giving 99.9% humidity for a long time, so I am trying to replace it with a new one.
However, when I connect it the sensor gives no reading and the debug log says:

## Sending to sensor.community - SDS011
Succeeded - api.sensor.community
DHT11/DHT22 read failed
DHT11/DHT22 read failed
DHT11/DHT22 read failed
DHT11/DHT22 read failed
DHT11/DHT22 read failed

What to do? I have already been toying with it for a while, checking the wire order etc etc. Would be very happy with any suggestions!

Maarten
Amsterdam

ricki-z, 2022-01-18:
Hello could you please check that the cables are properly connected to the pins of the DHT22?
The pins are really thin so it may help if you bent the pins a little.

MJLEVER00, 2022-01-22:
Thank you, Ricki-Z. 
I have bent the pins, but it makes no difference. I tried different cables: no difference.
I bought yet another DHT22: same result.
I bought a BME280 hoping it would work: same result (no readings...)
What to do? My room is littered with sensors, I am a bit desperate.
Can it be that I somehow need to "reset" the software in order to forget that old DHT sensor with its forever stuk 99.9% humidity reading? If not, what ELSE could it possible be?

pjgueno, 2022-01-22:
If you did not connect well at first you can have burnt the pins. I have already seen it.

When you tried with the BME280, did you change the config in the WiFi interface ? Did you use the right pins for SCL and SDA ?

MJLEVER00, 2022-01-23:
Thanks Pjgueno,
This is how I connected the wires:
SDA -> D3
SCL -> D4
GND -> G
VCC -> 3V
pins CSB and SDO are unused.

Burning the pins: how would I recognise that? I would say I am rather precise and never mis-connected them, but who knows.

Would it make sense to erase the firmware and start the installation all over? Could that make any difference? So far I didn't do that because I was afraid it would somehow break the link with my old sensor data.

Any suggestions much appreciated!

pjgueno, 2022-01-23:
**Did you select BME instead of DHt22 in the config ?**

[image]

Reflash should not be necessary but anyway, you won't "break the link". The board ID is generated from the MAC address of the ESP. It will be the same and actually the config file is persistent.

MJLEVER00, 2022-01-23:
Yes, I did definately select the BME280 sensor...
Here is with I get:

[image]

(It does show up as a BM**P**280 sensor though, might be that it only recognises the BME once it gets its first reading?)

And this is with both new sensors attached:

[image]

I did notice, by the way, that each DHT22 sensor that I have (I now have 3 of them) looks slightly different. The one is a bit thicker, the other has smaller holes in the casing, etc etc. Could it be that I still haven't found the precise DHT22 type that DOES work? (But then again, what about the BME280?)

MJLEVER00, 2022-01-23:
Additionally, here are the debug logs for both the DHT22 and the BME280 sensors (for the sake of clarity, only 1 sensor connected each time):

[image]

[image]

I have no idea what to do next. Again, suggestions most welcome :)

pjgueno, 2022-01-23:
It is not normal, that the BME280 is recognized as a BMP280 in such a case.
Try to reflash and send me some  screenshot of each config tab. You are doing something wrong but I can't imagine what...

How come you are sending something to madavi right after the BMx Test ?

[image]

MJLEVER00, 2022-01-24:
Hello Pjgueno,
thanks a lot for your questions.
I did reflash, below are screenshots of all the relevant config tabs. This time with both sensors connected... Same situation... No readings. Suggestions?

Kind regards
Maarten
Amsterdam

(By the way: I have no clue what it is sending if it has no sensor data. I am not familiar with the code...)

[image]
[image]
[image]
[image]
[image]
[image]

stewecar, 2022-01-24:
Hello,
did you check the esp8266 3,3V pins voltage? did you check the cables in continuity mode with a multimeter? I had once a lot of damaged cables that did not work well. Check also joint quality and make new welding joints if pins are corroded or not giving enough voltage. Lemme know. Cheers!

MJLEVER00, 2022-01-28:
Thanks for the suggestion!
Tried it. Stable 3.3 V over both the pins AND the welding joints.
It's so strange...

MJLEVER00, 2022-01-28:
Solved it (at least most of it).
Bought a new NodeMCU, new shorter cables and whereas both the new DHT22 sensors still don't work, the BME280 does! And now it also appears as a BME280 (no longer as a BMP280).
Thanks, Pjgueno and Stewecar, for your support.
My sensor can go back outside.

closed 2022-01-28 by MJLEVER00