opendata-stuttgart/sensors-software #943: ESP32

Beppi4U, 2022-01-17:
Hi there!
Not really an issue......but... 
Is ESP32 platform supported?  
And if "Yes" are the benefits significant? Where's to find the build Tut's?(i guess it should be the same)   
I mean besides the speed of the system/chip memory etc. the WiFi range should be greater than compared to ESP8266!?
Is the future platform ESP32?
Thanks 
Andreas

ricki-z, 2022-01-18:
At the moment the ESP32 code is very beta and may not really compile or even work.
One of the benefits of the ESP32 would be the hardware serial ports. So the SDS011 and a GPS module should work at the same time (the ESP8266 has some problems as UART is simulated in software).
And there are ESP32 modules with integrated LoRaWAN. This would make it possible to build sensors without the need for a wifi network that is always on.

pjgueno, 2022-02-01:
I have already succeeded to flash the beta on a ESP32 do it dev kit, on a TTGO Paxcounter and on a Heltec Wifi Lora 32. A BME 280 and multiple types of sensors were plugged on. Just take care of the pinout in the ext-def.h file.

It works quite flawlessly except for the https call of the Madavi and SC API which is not supported on ESP32. Just let the checkboxes unchecked. 

You can then register the ESP32 with the HexID (the one with letters within) in devices.sensor.community. Of course remove the auto update!

About the LoRaWAN, I succeed to make an Heltec Wifi LoRa 32 work in Helium and multiple TTN sensors already exist. THe decoder and integration can be written straight forward.

jfsgithub2020, 2022-03-06:
My ESP32 is running airrohr FW for over a year now. Sometimes a little unstable, but that got better when I replaced the USB power supply ;-) - as it has been a while, I don't remember exactly what I needed to fix, but it was kind of obvious when the compile errors show up. It had to do with compatibility of libraries, so it might depend on your grown environment. I use a regular SDS011 and a BME280.

AlexNikolovLON, 2023-05-07:
Hi, the beta esp 32-v2

compiles fine on PIO
but when Sending to opensensemap:
getting error Client sent an HTTP request to an HTTPS server.

## Sending to opensensemap:
Request http failed with error: 400
Details:Client sent an HTTP request to an HTTPS server.

Phaze-III, 2023-05-08:
@AlexNikolovLON could you try with this patch:

```diff
diff --git a/airrohr-firmware/airrohr-firmware.ino b/airrohr-firmware/airrohr-firmware.ino
index 930c5a7..792f96a 100644
--- a/airrohr-firmware/airrohr-firmware.ino
+++ b/airrohr-firmware/airrohr-firmware.ino
@@ -5695,7 +5695,7 @@ static unsigned long sendDataToOptionalApis(const String &data)
        {
                debug_outln_info(FPSTR(DBG_TXT_SENDING_TO), F("opensensemap: "));
                String sensemap_path(tmpl(FPSTR(URL_SENSEMAP), cfg::senseboxid));
-               sum_send_time += sendData(LoggerSensemap, data, 0, HOST_SENSEMAP, sensemap_path.c_str(), false);
+               sum_send_time += sendData(LoggerSensemap, data, 0, HOST_SENSEMAP, sensemap_path.c_str(), true);
        }
 
        if (cfg::send2fsapp)
```