opendata-stuttgart/sensors-software #950: issue with BMP280 data uploading

RobyGitH, 2022-02-02:
hi, I recently replaced BME280 with BMP280 (due to the high cost and low longevity). The sensor is recognised, along with the SDS011 in my Particulate matter sensor and visible at 192.168.1.X. I can see data for PM, temperature and pressure correctly.
However, while PM data from SDS011 are correctly uploaded onto https://maps.sensor.community, data from BMP280 are not (no sensor visible on the map). Same issue with data uploading onto https://opensensemap.org (only PM data)
Any idea how i can fix this?
thanks in advance.

ricki-z, 2022-02-14:
Hello,
did you change the config of your device at https://devices.sensor.community/?
You may also need to change the settings at opensensemap.org. This is a project completely independent from sensor.community so you may ask there what you need to change.

RobyGitH, 2022-02-14:
hi, thanks for the help. I have now changed the configuration and temperature and pressure are uploaded correctly. However, the map shows only PM 2.5, PM10, temperature but does NOT show pressure. Any reasons for this? 

From a look on the map, it looks like that the field 'pressure_at_sealevel' is used for BME280, but this is not available for my BMP280 (only the field 'pressure' is available). Can this be the reason why my sensor pressure is not mapped? If so, this could be a bug. If not, what could be the reason?

thanks!

ricki-z, 2022-02-15:
Could you please send me the chipID of your device to tech (at) sensor.community? I will look at this.
The field pressure_at_sealevel should also be available for the BMP280, but it's only created for sensors where we have a height above sea level. So with the chipID I can check if there was any problem with the lookup of this value.

RobyGitH, 2022-02-15:
hi,
the sensors are:
29700 PM
29701 temperature and pressure
However, I checked now on the map and they are finally showing correctly.
For some reasons, data for pressure started being recorded at 23:02, while
for temperature 40 minutes earlier, at 22:20. Hence, last night I thought
data for pressure was not recorded at all, but it was simply delayed. It
seems strange as they are on the same sensor and both temperature and
pressure were visible on the cloud under 29701.
Thanks again for your help!

ricki-z, 2022-02-15:
Maybe there was a problem with reading the correct pressure values at start up. The map is filtering unrealistic values. As the values became more realistic the map showed them.

RobyGitH, 2022-02-15:
I checked the data for yesterday and today, and yesterday the field "pressure_at_sealevel" was not present, today it is. This is probably because the altitude is now 40, while yesterday it was null. This probably caused the issue with the map.
Anyway, it seems to be working well now, thanks again for your help.

closed 2022-02-15 by ricki-z