opendata-stuttgart/sensors-software #951: Problems with I2C - DNMS sensor not found

imaluschi, 2022-02-04:
Update of firmware of existing (and working) Sensors makes problems, the DNMS is not found anymore.
## Configuration
- ESP8266 with SDS011, BME280 and DNMS (without P82B715, connection cable about 150mm).
- DNMS with Teensy 4.0 updated to Firmware V2.1.4
- airrohr Firmware NRZ-2021-134-B4, downloaded from github on 01.02.2022
- Firmware built with Arduino 1.8.19
## Result
SDS and BME280 is working, but DNMS sensor is not found. So only one I2C-device is found. Cabling is ok, was working before update. Problem persists after using older firmware versions and other esp8266-core versions.
[image]
## Tries
- DNMS all versions tried (V1.0.5, V2.0.4, V2.0.5, V2.1.4, V2.1.5) with either USB-Type Audio or Serial+MIDI+Audio
- airrohr NRZ-2021-134-B4 and stable version NRZ-2020-133
- Building of firmware with esp8266-core Version 2.6.2, 2.7.3 and 3.0.2
## Investigations
Upload of an I2C-Scanner to ESP8266: https://gist.github.com/jmas/01280e8c1aae8060c90c029e59647cd9
Both devices where found (BME280 on 0x76 and DNMS on 0x55H). This also indicates the wiring is working.
[image]
Strange.
So took the code from the I2C-Scanner loop() and inserted it in airrohr-firmware.ino. Therefore defined a new subroutine
`void i2cScanner()`
Called this sub in the sub setup on different places (in void setup(void) ).
Relevant positions:
```
i2cScanner();
Debug.println("powerOnTestSensors()");
	powerOnTestSensors();
	logEnabledAPIs();
	logEnabledDisplays();
i2cScanner();
	delay(50);
	starttime = millis(); // store the start time
```
## Finding
Both devices are found before powerOnTestSensors(), DNMS is not found afterwards.
[image]
Seems there is a problem occuring during
`initDNMS() or dnms_reset()`

I'm not able to investigate deeper.
Idea, anyone?

pjgueno, 2022-02-04:
Thanks a lot for you complete description !

I am on the NRZ-2020-133/DE firmware for my DNMS:
[image]
But the Teensy was not updated for a long time.

Can you try to flash the latest.bin with the flasher ?

I hope it is not something with the DNMS code. In that case can you also open an issue here?  [https://github.com/hbitter/DNMS/issues](https://github.com/hbitter/DNMS/issues)

Can you also check for interferences in the I2C line to the DNMS. It is a reported issue. How long is your cable ? Do you have an extension bus ?
 
I will check the PowerOntestsensor() function. I don't think something changed in it for a longtime.

imaluschi, 2022-02-04:
No extension bus (as described), cable about 150mm long. Can't imagine interferences, the cable is shielded, connection with plug (XH2.54) on both sides.
Tried also to unplug the BME280, as expected DNMS is the only I2C device detected before powerOnTestSensors(); and not detected afterwards.
DNMS was additionaly plugged to Laptop during the tests, therefore not rebooted because of plugging/unplugging to/from power.
As written, it is detected before powerOnTestSensors() and not afterwards, no change during this time on the DNMS.

airrohr-firmware was also updated via AutoUpdater, to stable version NRZ-2020-133 and to beta version NRZ-2021-134-B4 - I think that will make flashing the bin obsolete
[image]

[image]
.

imaluschi, 2022-02-05:
Inserted the I2C-Scanner into powerOnTestSensors()
```
if (cfg::bmx280_read)
{
     Debug.println("powerOnTestSensors(), BMX280 ( if (cfg::bmx280_read) )");
     i2cScanner();  
		debug_outln_info(F("Read BMxE280..."));
		if (!initBMX280(0x76) && !initBMX280(0x77))
		{
			debug_outln_error(F("Check BMx280 wiring"));
			bmx280_init_failed = true;
		}
     Debug.println("powerOnTestSensors(), BMX280 Ende )");
     i2cScanner();
}
```
[image]
Seems ok.

The DNMS is always found in 0x55, but not reconized after initDNMS(). A short time later, after finishing powerOnTestSensors() ist is recognized again.
```
if (cfg::dnms_read)
{
      Debug.println("powerOnTestSensors(), DNMS ( if (cfg::dnms_read) )"); 
      i2cScanner();  
		debug_outln_info(F("Read DNMS..."));
		initDNMS();
      Debug.println("powerOnTestSensors(), DNMS Ende )");  
      i2cScanner();  
	}
}
```
[image]

Could there also be an issue because the I2C-Scanner finds the DNMS under 0x55 not under 0x55H ?

imaluschi, 2022-02-05:
Seems dnms_reset() is making problems. Commented it out, and correct DNMS-version is found.

```
static void initDNMS()
{
	char dnms_version[DNMS_MAX_VERSION_LEN + 1];

	debug_out(F("Trying DNMS sensor on 0x55H "), DEBUG_MIN_INFO);
//	dnms_reset();
	delay(1000);
	if (dnms_read_version(dnms_version) != 0)
	{
		debug_outln_info(FPSTR(DBG_TXT_NOT_FOUND));
		debug_outln_error(F("Check DNMS wiring"));
		dnms_init_failed = true;
	}
	else
	{
		dnms_version[DNMS_MAX_VERSION_LEN] = 0;
		debug_outln_info(FPSTR(DBG_TXT_FOUND), String(": ") + String(dnms_version));
	}
}
```
[image]

Seems to work without dnms_reset() !
[image]

imaluschi, 2022-02-08:
Works so far without problems.
[image]
.

pjgueno, 2022-10-11:
Force dnms_reset()

closed 2022-10-11 by pjgueno