opendata-stuttgart/sensors-software #953: NRZ-2020-133 boot loop

msebald, 2022-02-14:
Hello all,

after trying to uncheck DHT22 in settings of my sensor it did not boot anymore. So i reinstalled the firmware and also hooked it up to my serial connection. What I see after it was upgrading (or downgrading?) is this:

```
airRohr: NRZ-2020-133/DE
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to FlashForward
....SNTP synced: Mon Feb 14 09:28:13 2022

twoStageOTAUpdate
output debug text to displays...
.
WiFi connected, IP is: 172.27.0.191

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd30 end: 3fffffc0 offset: 0190
3ffffec0:  00000003 00000000 3ffef4b0 3fffff10  
3ffffed0:  3fffff48 00000050 3fffff10 3ffef27f  
3ffffee0:  3ffe8939 3ffe8934 3ffef4b0 40218194  
3ffffef0:  3ffe892b 3ffe8930 3ffef4b0 4020ab58  
3fffff00:  00000040 00000000 feefeffe feefeffe  
3fffff10:  70747400 000c0000 80201b44 402013cc  
3fffff20:  feefeffe feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe feefeffe  
3fffff40:  feefeffe feefeffe 00706300 bf001bac  
3fffff50:  80efeffe 00000000 00000000 feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe 01004544 feef010d feefeffe  
3fffff80:  3ffeea31 feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffef6a8  
3fffffa0:  3fffdad0 00000000 3ffef668 40223240  
3fffffb0:  feefeffe feefeffe 3ffe8690 401006c9  
<<<stack<<<
```

What can I do? I noticed that I only can update to the stable firmware through the Windows updater tool.

msebald, 2022-02-14:
Same when I install the latest beta via esptool.exe:

```
airRohr: NRZ-2021-134-B4/DE
mounting FS...
opened config file...
parsed json...
output debug text to displays...
Connecting to xxx
.....
WiFi connected, IP is: xxx.xxx.0.191

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Exception (28):
epc1=0x4000bdc8 epc2=0x00000000 epc3=0x00000000 excvaddr=0x00000000 depc=0x00000000

>>>stack>>>

ctx: cont
sp: 3ffffd30 end: 3fffffc0 offset: 0190
3ffffec0:  00000050 00000001 00000000 08600800  
3ffffed0:  7fffffff 00000003 3fffff48 3ffef7d2  
3ffffee0:  3ffe8af5 3ffe8afd 3ffef810 402181b8  
3ffffef0:  3ffe8af1 3ffe8aec 3ffef810 4020cf18  
3fffff00:  00000040 00000000 feefeffe feefeffe  
3fffff10:  00000000 000c0000 00000000 402013a0  
3fffff20:  feefeffe feefeffe feefeffe feefeffe  
3fffff30:  feefeffe feefeffe feefeffe feefeffe  
3fffff40:  feefeffe feefeffe 00000000 bf001bac  
3fffff50:  00000000 00000000 00000000 feefeffe  
3fffff60:  feefeffe feefeffe feefeffe feefeffe  
3fffff70:  feefeffe feefeffe 4544effe 010d0100  
3fffff80:  feefeffe feefeffe feefeffe feefeffe  
3fffff90:  feefeffe feefeffe feefeffe 3ffefa68  
3fffffa0:  3fffdad0 00000000 3ffefa54 40224a34  
3fffffb0:  feefeffe feefeffe 3ffe87a8 4010077d  
<<<stack<<<
```

ricki-z, 2022-02-14:
This looks like a problem with the config settings.
You should delete the config also with the flashing tool and reconfigure it.

msebald, 2022-02-14:
> This looks like a problem with the config settings. You should delete the config also with the flashing tool and reconfigure it.

Okay. But what about my sensor ID? Will it be lost or not? If yes, can it be restored?

pjgueno, 2022-02-14:
> > This looks like a problem with the config settings. You should delete the config also with the flashing tool and reconfigure it.
> 
> Okay. But what about my sensor ID? Will it be lost or not? If yes, can it be restored?

Don't worry. It is generated from the MAC address. It will stay the same. You'll just have to check the config.

msebald, 2022-02-15:
> Don't worry. It is generated from the MAC address. It will stay the same. You'll just have to check the config.

Sensor works again, thank you. And it is the same ID. Thank you also for making this clear.

After checking the beta checkbox I also got the beta firmware. Perfect.

I noticed that I cannot set the altitude above sea level as it is always 0 again after the sensor is restarted. Also I need to check why I only see the DHT22 data in Sensor.Community map and no data (including pressure) from the BME280 module. And it is not clear for me what temperature and humidity data I see in the Luftdaten addon for openHAB when both modules are active. But this has nothing to do with this issue here.

closed 2022-02-15 by msebald