opendata-stuttgart/sensors-software #956: sensor BMP presion atmosferica fuera de rango o incorrecta

Krisdragow, 2022-04-02:
### ID: 7965572 (500291798b84)
Versión del Firmware: NRZ-2020-133/ES (Nov 29 2020)

Pressure (hPa): 82068.00 
Temperature (°C): 23.00
----
Sending to sensor.community - BMP180
Succeeded - api.sensor.community
Sending to madavi.de: 
Succeeded - api-rrd.madavi.de
 Sending to Server FS App: 
Succeeded - server.chillibits.com
 Sending to aircms.online: 
Request failed with error: 403

Details:
 Sending to custom influx db: 
Sending to custom api: 
Sending as csv: 
CSV Output: {"software_version": "NRZ-2020-133", "sensordatavalues":[{"value_type":"BMP_pressure",

pjgueno, 2022-04-02:
It is the serial log not the values which are sent.
We still have not corrected hPa to Pa...
It does not matter actually:
[image]

But maybe you mean the 820 hPa...Are you in the middle of a tornado ? Or do you live at 2000 m altitude ?

Krisdragow, 2022-04-06:
_But maybe you mean the 820 hPa...Are you in the middle of a tornado ?_  haha it seems like it lol

i done this, just reinstall all the firmware and it works good i think... 
but i forgoten to say, i use the pms5003 and not the sds011, is necesary tu conect RX and TX (PMS5003)?

pjgueno, 2022-04-07:
You can connect like the sds011.
Check the datasheet of the pms5003 to see the pinout.

Don t forget to select the right sensor everywhere.

Krisdragow, 2022-04-08:
thx a lot, now all is working good.

closed 2022-04-08 by Krisdragow