opendata-stuttgart/sensors-software #957: Wifi not unencrypted - where to find password?

Troubadix, 2022-04-27:
Hi,

I downloaded the latest_de.bin Beta and flashed it using esptool (because your flashing tool refuses to work because of your expired SSL certificate).
I can see the Wifi network the sensor provides, but it requires an password.
Your documentation says it is an unecrypted wifi - so where can I find the password for the wifi?

pjgueno, 2022-04-27:
You should not have downloaded the beta...
The PW is airrohrcfg.

Troubadix, 2022-04-27:
Thank you.
The final is about 10000 years old, even the beta is way too old. 
I'm using the beta with my old sensor, so no need for this grandma-final :)
Is this project still in development?

pjgueno, 2022-04-27:
Of course it is still in development.

What do you mean old ? I pushed in the beta a few weeks ago. 

The master works well.

Troubadix, 2022-05-10:
My sensor reports version "NRZ-2021-134-B4/DE (Aug 22 2021)" as the latest beta? How does this fit with "a few weeks ago"?

ricki-z, 2022-05-17:
There is a difference between the latest beta compiled and available as download and the source code.
We don't compile a new version for each change in the source code (i.e. beacause we need to check it first before we make it available for sensors with beta versions installed).

pjgueno, 2022-10-11:
Questions answered

closed 2022-10-11 by pjgueno