opendata-stuttgart/sensors-software #962: CCS811 Sensor Support

hristodobrev, 2022-08-26:
Hello,

Are you looking forward to support the Carbon Dioxide Detection Sensor (CCS811)?

Currently, there are PM, Temp, Humidity, Air Pressure options only.

If not, is there a way to edit the source code myself (if there is an Arduino version which I could flash later on my NodeMCU)?

[image]

Thanks,
Chris

pjgueno, 2022-08-26:
Thanx!
There is the SCD30 in the beta version.
There is still an issue with calibration and the project was firstly aimed at outdoor.
You can do a PR in the beta branch. For another project I implemented MHZ 16 and 19.

pjgueno, 2022-08-26:
For thé development please use platformio. It is easier as it automatically load the libs.
Be sure to git clone the current beta branch when you starts.

hristodobrev, 2022-08-26:
Thanks!

I will give it a play these days and will share a feedback afterwards.