opendata-stuttgart/sensors-software #963: airrohr-firmware: Missing dependencies SSD1306.h and SH1106.h

lcewcb8l6xe1jxr0zxdj, 2022-09-01:
Hi,

I unfortunately cannot build airrohr-firmware due to missing dependencies to SSD1306.h and SH1106.h.
While building, PlatformIO cannot resolve dependency to ESP8266_SSD1306@4.1.0 (platformio.ini):

```
Library Manager: Installing ESP8266_SSD1306 @ 4.1.0
Warning! Could not find the package with 'ESP8266_SSD1306 @ 4.1.0' requirements for your system 'linux_x86_64'

```
which leads to
```
/home/xyz/git/sensors-software/sensors-software/airrohr-firmware/airrohr-firmware.ino:99:21: fatal error: SSD1306.h: No such file or directory
```

I am trying for hours, please give me a hint if I missed something. I held on to this article: https://github.com/opendata-stuttgart/meta/wiki/Firmware-einspielen, "Variante 1 "PlatformIO und VSCode"".
I also tried in Windows, same problem. 

Thank you in advance!

ricki-z, 2022-09-01:
Hello,
the error is telling that you try to build the firmware for a  x86 Linux system. But the sensor is using an ESP8266. Maybe you haven't installed the ESP8266 extension for platform.io.
Why don't you try to flash the firmware with our flashing tool? You can find the up to date install instructions at https://sensor.community/en/sensors/airrohr/

lcewcb8l6xe1jxr0zxdj, 2022-09-05:
Hi,
thank you for your quick answer. 

After a lot of trial-and-error I found out, this PIO-Library had to be installed manually: 
`"ESP8266 and ESP32 OLED driver for SSD1306 displays by ThingPulse"`

It can be found in "Libraries" in PIO in VSCode by searching for
`"ESP8266 and ESP32 OLED driver"` - the full string leads to no result.  

I'd like to enable the firmware for InfluxDB Cloud, which takes an API-Key instead of user/password authenticaiton.

Thank you! :)

closed 2022-09-05 by lcewcb8l6xe1jxr0zxdj